candle-usb/candleLight_fw #2: Multiple device support

nexulm, 2016-11-29:
The firmware seems not to support multiple candleLight, cantact, CANable interfaces plugged in at the same time.
If I'm using the candleLight build and plug in two USB2CAN devices socketcan (i.e. ip link set can0 up type can bitrate 500000) is coming up with two CAN interfaces (can0 / can1) but in reality only one device is working.

Is there any workaround available?

strupppi, 2016-11-29:
Hi,
the gs_usb kernel module had a bug in some older kernel releases. One candleLight device claimed two CAN interfaces (can0 AND can1). Only the of them first one worked.

If you plug in two candleLight devices, can you check if you get four CAN interfces, i.e. CAN0 to CAN3?
In that case, can0 and can2 should work.
This problem is fixed in newer kernel releases

nexulm, 2016-11-29:
I'm using Kernel 4.4.32+on a Raspi B+ and that was a good hint. I've never seen before thate setting up can 0 and can2 is an option. But the two candleLight devices are running fine now.
Two additional questions:
1) Does https://github.com/HubertD/socketcan_gs_usb includes the fix?
2) If I'm using the gs_usb module from 1) or a newer kernel >4.5.x what are the limitations of the amount of candleLight devices? Is it possible to use 4 of them at the same time?

strupppi, 2016-11-29:
1. yes, the fix is included there
2. i'm not sure what the limits are, but 4 of them should work without problems

closed 2017-02-24 by HubertD