candle-usb/candleLight_fw #3: Nonfunctional with kernel 4.9+

normaldotcom, 2017-02-23:
Have you attempted to use the candleLight with kernels 4.9 and later? I get a stack trace and error when running on these kernels. I have tested across multiple hardware versions and kernel versions, and the problem occurs beginning with kernel 4.9.

```
[   43.354559] usb 2-2: new full-speed USB device number 3 using ohci-pci
[   43.666367] CAN device driver interface
[   43.667960] ------------[ cut here ]------------
[   43.667967] WARNING: CPU: 1 PID: 872 at drivers/usb/core/hcd.c:1584 usb_hcd_map_urb_for_dma+0x42a/0x550 [usbcore]
[   43.667968] transfer buffer not dma capable
[   43.667970] Modules linked in: gs_usb(+) can_dev cfg80211 rfkill crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul snd_intel8x0 glue_helper ablk_helper snd_ac97_codec cryptd ppdev snd_pcm psmouse i2c_piix4 input_leds snd_timer e1000 video fjes intel_agp intel_gtt joydev intel_cstate intel_rapl_perf snd soundcore mousedev ac97_bus led_class pcspkr mac_hid battery ac parport_pc parport evdev button acpi_cpufreq tpm_tis tpm_tis_core tpm sch_fq_codel vboxvideo(O) ttm drm_kms_helper drm syscopyarea sysfillrect sysimgblt fb_sys_fops vboxsf(O) vboxguest(O) ip_tables x_tables ext4 crc16 jbd2 fscrypto mbcache hid_generic usbhid hid sr_mod sd_mod cdrom ata_generic pata_acpi ohci_pci serio_raw atkbd libps2 ehci_pci ahci ata_piix libahci ehci_hcd ohci_hcd libata
[   43.668009]  usbcore usb_common scsi_mod i8042 serio
[   43.668014] CPU: 1 PID: 872 Comm: systemd-udevd Tainted: G           O    4.9.11-1-ARCH #1
[   43.668016] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[   43.668017]  ffffc90005333760 ffffffff81305830 ffffc900053337b0 0000000000000000
[   43.668020]  ffffc900053337a0 ffffffff8107eb3b 0000063000000000 ffff8801fab356c0
[   43.668022]  0000000000000000 ffff8802157ff800 0000000000000001 ffff8802141cb000
[   43.668025] Call Trace:
[   43.668029]  [<ffffffff81305830>] dump_stack+0x63/0x83
[   43.668032]  [<ffffffff8107eb3b>] __warn+0xcb/0xf0
[   43.668034]  [<ffffffff8107ebbf>] warn_slowpath_fmt+0x5f/0x80
[   43.668039]  [<ffffffffa00a189a>] usb_hcd_map_urb_for_dma+0x42a/0x550 [usbcore]
[   43.668042]  [<ffffffff8118a9f5>] ? __alloc_pages_nodemask+0xf5/0xef0
[   43.668045]  [<ffffffffa00a28c5>] usb_hcd_submit_urb+0x335/0xb30 [usbcore]
[   43.668047]  [<ffffffff810b5d7a>] ? update_curr+0x6a/0x190
[   43.668051]  [<ffffffffa00a4274>] usb_submit_urb+0x2f4/0x560 [usbcore]
[   43.668053]  [<ffffffff813061f2>] ? get_from_free_list+0x42/0x50
[   43.668056]  [<ffffffffa00a4a1e>] usb_start_wait_urb+0x6e/0x170 [usbcore]
[   43.668060]  [<ffffffffa00a4bfc>] usb_control_msg+0xdc/0x120 [usbcore]
[   43.668063]  [<ffffffffa038922b>] gs_usb_probe+0x7b/0x5b0 [gs_usb]
[   43.668065]  [<ffffffff8145e6f0>] ? __pm_runtime_set_status+0x1c0/0x270
[   43.668076]  [<ffffffffa00a9459>] usb_probe_interface+0x159/0x2d0 [usbcore]
[   43.668079]  [<ffffffff81452713>] driver_probe_device+0x223/0x430
[   43.668081]  [<ffffffff814529ff>] __driver_attach+0xdf/0xf0
[   43.668083]  [<ffffffff81452920>] ? driver_probe_device+0x430/0x430
[   43.668085]  [<ffffffff8145024c>] bus_for_each_dev+0x6c/0xc0
[   43.668088]  [<ffffffff81451e5e>] driver_attach+0x1e/0x20
[   43.668090]  [<ffffffff81451880>] bus_add_driver+0x170/0x270
[   43.668092]  [<ffffffff814533c0>] driver_register+0x60/0xe0
[   43.668095]  [<ffffffffa00a7fa1>] usb_register_driver+0x81/0x140 [usbcore]
[   43.668097]  [<ffffffffa007c000>] ? 0xffffffffa007c000
[   43.668100]  [<ffffffffa007c01e>] gs_usb_driver_init+0x1e/0x1000 [gs_usb]
[   43.668102]  [<ffffffff81002190>] do_one_initcall+0x50/0x180
[   43.668105]  [<ffffffff8117f6ee>] do_init_module+0x5f/0x1f1
[   43.668108]  [<ffffffff8110d364>] load_module+0x2384/0x2a50
[   43.668110]  [<ffffffff8110a240>] ? symbol_put_addr+0x50/0x50
[   43.668113]  [<ffffffff811cb32a>] ? vmap_page_range_noflush+0x25a/0x350
[   43.668115]  [<ffffffff8110dba4>] SyS_init_module+0x174/0x190
[   43.668117]  [<ffffffff81003b64>] do_syscall_64+0x54/0xc0
[   43.668119]  [<ffffffff8160abab>] entry_SYSCALL64_slow_path+0x25/0x25
[   43.668145] ---[ end trace 9f8b0186c8e48867 ]---
[   43.668149] gs_usb 2-2:1.0: Couldn't send data format (err=-11)
[   43.668155] gs_usb: probe of 2-2:1.0 failed with error -11
[   43.668216] usbcore: registered new interface driver gs_usb

Trace from kernel 4.9.11-1-ARCH
Reproduced on VirtualBox VM
Reproduced on 2013 Macbook Pro
```

normaldotcom, 2017-02-23:
I just tested this on an ARM platform (Odroid XU4 running Arch Linux ARM at 4.9.9-1-ARCH), and could not reproduce this problem. Looks like it could be something x86_64/DMA specific?

normaldotcom, 2017-02-24:
Turns out this is an issue with the gs_usb kernel module. Fixed on my development machine; working on a patch...

HubertD, 2017-02-24:
Sorry for the late answer. I didn't try 4.9 yet so I wasn't aware of the bug.
Thanks for reporting this and also many, many thanks for submitting your patches for socketcan!
Since it's a gs_usb problem, I'm going to close this issue?

normaldotcom, 2017-02-24:
Glad I could help! For reference, the patch can be found [here](http://marc.info/?l=linux-can&m=148795367315850&w=2).

closed 2017-02-24 by normaldotcom