candle-usb/candleLight_fw #4: WinUSB driver support

normaldotcom, 2017-05-24:
Have you attempted to write a driver for candleLight on Windows? I'm wondering how easy it would be to write a Python library that interfaces with the WinUSB device to add candleLight to python-can on Windows. Any suggestions?

`https://github.com/hardbyte/python-can/`

HubertD, 2017-05-24:
I implemented a WinUSB driver here:
https://github.com/HubertD/cangaroo/tree/master/src/driver/CandleApiDriver/api
It's far from perfect, but it works, and you should be able to bind to the dll from python.

I intend to write a better api in the near(er) future and maybe I will extend my gs_usb implementation to improve the windows performance.

I also plan to build a system where multiple windows applications can share one candlelight, maybe via named pipes...

So, I would be very glad to see a python-can driver for the candleLight and the upcoming *drumroll* candleZero. But please be prepared to make adjustments once the new API is there.

Closing this as I don't see it as an issue.
Please feel free to email me via github@cangaroo.org for any support you would need.

closed 2017-05-24 by HubertD

normaldotcom, 2017-05-24:
Excellent, I'll take a look at your existing API and keep you posted via email.

Nice work on the candleZero, I pulled down your hardware repo and the size is quite impressive! Do you plan on integrating it into a cable assembly?

HubertD, 2017-05-24:
exactly :-)
[image]
Very first, but working prototype