candle-usb/candleLight_fw #6: Multiple can interfaces showing up

oxoocoffee, 2017-10-27:
Hello,
I updated my two CANtact with your latest firmware and all looks good except when I plug in one of the boards to my machine (Linux Mint 18) I get can0 and can1  while one board is plunged in.
If I plug another board at the same time (two boards are plunged in) I see can0, can1, can2 and can3.
Did you ever encountered this? I can make can0 work just fine but the other board I am never sure if it is can1 or can2

strupppi, 2017-10-31:
It's a known issue with older kernel versions.
The firmware creates 2 USB interfaces, one for CAN, the other's meant for DFU software updates. Unfortunately, this wasn't handled properly by kernel drivers back then, so the kernel treated both USB interfaces as CAN interfaces. It's fixed now (since 4.9 IIRC, can't find the relevant commit ATM).

oxoocoffee, 2017-11-01:
Thank you for your response. I ungraded to  4.10.0-38 and problem is gone