candle-usb/candleLight_fw #8: Cantable clone fw

riodda, 2018-02-05:
Hi i've made a canable clone and i've tryed to flash both the binaries avilable but i have no usb device showing up nor any led activity, i guess that the problem is that canable has StM32F04 with internal oscillator, while candlelight board has F07 internal oscillator and cantact has F04 external oscillator, i guess in the makefile there is the need of a new board definition ? Sorry i'm not very much into making firmwares.
Thanks.

riodda, 2018-07-04:
Thanks, Much apreciated.

Entropy512, 2019-04-22:
@riodda Did nexulm's fork build/start correctly for you?

An alternative possibility may be a GCC issue.  I've found that nexulm's fork (which has additional patches for canable not in this repo) behaves exactly as you describe if built using the gcc-arm-none-eabi present in Ubuntu 18.04 or the team-gcc-arm PPA, but works fine if built on an ubuntu 16.04 machine.

I haven't tried this specific repo.

riodda, 2019-04-23:
It's long time i don't work on this, but it's a good clue i will try to make one board building with a 16.04 and see if it solvers the problem.
Cheers.

Entropy512, 2019-08-06:
@riodda Did you have any luck with 16.04?  If so it would be interesting to know what distro @HubertD and @normaldotcom use when building.

(I figure it's time to get back to cleaning up my patches to handle alternative host endianness, but it would be nice to figure out why I can only get functional builds on very old Ubuntu releases.)

riodda, 2019-08-06:
Sorry i haven't got the chance to test it, that project is a bit on hold.
Cheers.

fenugrec, 2021-02-16:
The build system has changed quite a bit since this was opened, is this still an issue ? Otherwise will close in a few weeks.

fenugrec, 2021-04-25:
Feel free to retest current firmware, and open issues as necessary.

closed 2021-04-25 by fenugrec