candle-usb/candleLight_fw #14: Wronge Controller in Makefile

Scytie, 2018-08-27:
Currently it is:

```
candleLight:
        $(MAKE) CHIP=STM32F042x6 BOARD=candleLight bin

flash-candleLight:
        $(MAKE) CHIP=STM32F042x6 BOARD=candleLight board-flash

cantact:
        $(MAKE) CHIP=STM32F072xB BOARD=cantact bin

flash-cantact:
        $(MAKE) CHIP=STM32F072xB BOARD=cantact board-flash

```
but it has to be vice versa. Cantact uses F042 and candelight uses F072.

fenugrec, 2019-10-29:
This seems true. I have neither hardware to test though, @HubertD  can you confirm this ?
If so, I can quickly patch the Makefile and eventually CMakeLists.txt also.

HubertD, 2019-10-30:
can confirm at least for the candleLight.
At least the one I've got here has a STM32F072.

HubertD, 2019-11-03:
fixed in current master

commit 90400f03 (2019-11-03): port fix #14
closed 2019-11-03 by HubertD