candle-usb/candleLight_fw #19: CAN able pro flashed with latest version of firmware no windows drivers

ve3sjk, 2019-06-29:
I recently received 2 new CANable devices and flashed the latest firmware using the webpage app on the canable website. The firmware flashes without error but when re insert the devices i get to usb devices under win7 without drivers. The first is gs usb and the 2nd is a programming port. Win7 reports that no drivers are installed and cangroo does not see and devices.

normaldotcom, 2019-07-03:
Did you attempt to restart Cangaroo after plugging in the device? The two devices that show up are correct; however, they should use the native WinUSB driver with no extra legwork.

fenugrec, 2019-11-11:
Closing this until more information or more questions are added. Feel free to reopen.

closed 2019-11-11 by fenugrec