candle-usb/candleLight_fw #20: Windows 10 - drivers do not work as expected

frogcrush, 2019-07-02:
When trying to use on Windows 10, I get the following showing up in Device Manager:

jhofstee, 2019-10-29:
You wrote "drivers do not work as expected", but you forget to add what you dud expect to happen..

fenugrec, 2020-11-01:
Just plugged in mine on a random win10 box and they were detected properly. I would close this as "works here" unless you still have issues ?

frogcrush, 2020-11-01:
I don't work at that place any more, so I have no way of testing or explaining any more unfortunately, so I agree that if it works for you then we can close this ticket :)

fenugrec, 2020-11-01:
Perfect ! thanks for the update @frogcrush .

stmlex, 2023-12-01:
Hi!
My Canable 2 adapter with candlelight_fw-ba6b1dd do not recognized on my computer
[image]

[image]

alexandration, 2023-12-01:
@fenugrec, thanks for your answer. 

Just maybe someone has a solution. 

This adapter worked on one computer running Windows, but after trying to find the reason in the regedit tool, I broke it too. And @HubertD dll worked perfectly.

fenugrec, 2023-12-04:
Thanks for the additional details @alexandration  .
Reopening since we may be doing something non-windows-friendly in our descriptors.  Unfortunately I don't have time to dig into this (and no win* machines). I think it will be hard to bisect for the F4 targets since they were added recently, also maybe this perhaps isn't a regression, I'm not sure. Canable were supported forever, it might be worth testing a build from ~2019 or so (after cleaning the registry as you described, to force the descriptor query).

I hope this isn't related to our use of unique serial numbers in one of the strings...

Lyubiao, 2024-01-27:
> @fenugrec 
> Please tell me, is it correct that the GS_USB interface descriptor index in the configuration descriptor is [zero](https://github.com/candle-usb/candleLight_fw/blob/master/src/usbd_gs_can.c?plain=1#L73)? Because when requesting a string descriptor, this interface has an index of [0xEE](https://github.com/candle-usb/candleLight_fw/blob/master/src/usbd_gs_can.c?plain=1#L635).
> I changed this value and the host started asking for a string descriptor and then a Microsoft Compatible ID Feature Descriptor.
> As a result, I have for my F4DiscoveryBoard:
> [image]
> and for canable2 adapter:
> [image]
> 
> And Savvy CAN works these adapters.
> 
> But before that I cleaned the registry `HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\1D50606F0000`.
> And removed the drivers of previous devices. After that I connected F4DevBoard with the corrected firmware. Since Windows requests the extended descriptor once and writes to the registry into usbflags, the canable2 adapter now works on my computer with factory firmware.

thinks，This method works

prounlocks, 2024-12-07:
> I recently did the same with this [project](https://github.com/normaldotcom/candleLight_fw) for the MKS CANable 2 (G431) adapter. But there it was also necessary to add `#define USBD_SUPPORT_USER_STRING_DESC 1` to the `usbd_conf.h` file. And everything works. Now I don't need F4DevBoard for the initial driver setup.
> 
> I think it's worth making a pull request with these fixes to the @normaldotcom project?

Got also MKS canable 2.0 and cant install drivers for both devices.  Used ZADIG tool, installed however - cangaroo not recognize my adapter with candlelight anymore.  BUSMASTER which they promised also dont have drivers.  I cant use this adapter in windows.

ziogianni92, 2025-04-17:
> [@fenugrec](https://github.com/fenugrec) Please tell me, is it correct that the GS_USB interface descriptor index in the configuration descriptor is [...]

I have the same issue, can you explain better how you solved?
I need F4DevBoard to do the same test or i can try without?

Thanks

Lyubiao, 2025-07-27:
Hello everyone, I've discovered a tool that can be used on Windows OS to clear USB devices. Run it as an administrator, select the problematic gs_usb devices, and delete them. Then, restart Windows, and it should be able to recognize the devices normally again.

[image]

- right-click and delete
[image]

[DeviceClean-V1.5.zip](https://github.com/user-attachments/files/21454328/DeviceClean-V1.5.zip)