candle-usb/candleLight_fw #29: [RFC] add git commit hash to USB serial / product string

fenugrec, 2019-11-05:
I'm working with 2 devices and jumping accross many branches, rebasing, etc, and I'm easily confused as to which build my devices are actually running.

I'm thinking it might be nice to add a short git hash (as produced by `git rev-parse --short HEAD` to one of the USB strings, but I'm not sure which one

- iSerial is already the built-in ID of the chip, and I think it's not the right place for a fw version anyway
- append to iProduct ? ugly, and irrelevant..
- add another USB string ID ? haven't looked into this, might be the way to go...
- some other mechanism ?
- is it even a good idea ?

HubertD, 2019-11-05:
unfortunately don't have a good idea where to put a software version, either.
it'd be nice to have, though.
Using iSerial is not a good option imho, as it would probably be used for udev rules to assign a certain can interface name to a certain can bus. That shouldn't change with a software update.
Using iProduct might not be great, but I see it as an option. At least, it would be easily accessible...

normaldotcom, 2019-11-06:
I append git rev/repo information to the product string on the stock CANable firmware. It's certainly not pretty, but it works well. Example dmesg output:

[1263955.892535] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1263955.892538] usb 1-1: Product: CANable 1205aa6 https://github.com/normaldotcom/cantact-fw
[1263955.892541] usb 1-1: Manufacturer: Protofusion Labs
[1263955.892543] usb 1-1: SerialNumber: 0024000D524E431120373533

brian-brt, 2020-12-02:
I'll add a :+1: for putting a git hash somewhere. A [vendor-specific descriptor](https://stackoverflow.com/questions/19464269/where-to-find-information-on-custom-usb-descriptor-types) which just has the string seems like a "clean" approach, but pretty annoying to retrieve.

It's been a while since I dealt with USB devices on Windows, but I think there's a possibility of it being unhappy if you change any of the standard strings without also changing bcdDevice. Ignoring the new value and caching the old one seems more unlikely. Have you done any experiments with that @normaldotcom?

fenugrec, 2020-12-02:
Just working on this right now ! Trying iConfiguration in the config descriptor (seems like the code was ready for this, config.h already had USBD_CONFIGURATION_STRING_FS so I just needed to add USBD_IDX_CONFIG_STR in the config descriptor).
It's pretty straightforward : lsusb -v will show the string that way. 

Interesting point re windows caching standard strings, although I can't see how they could assume that every usb gadget is guaranteed to change bcdDevice when they upgrade firmware ?

fenugrec, 2020-12-02:
According to https://techcommunity.microsoft.com/t5/microsoft-usb-blog/how-does-usb-stack-enumerate-a-device/ba-p/270685 (possibly outdated), it does mention caching for the  MS OS Descriptor Query ,

> subsequent enumerations of any device with the same VID/PID/Revision will read the bVendorCode from this registry value rather than querying the device. 

I think they have no choice (per USB) but to query the config descriptor every time. Just firing up a win7 VM to check.

fenugrec, 2020-12-02:
Ok. Tested in a win7 VM, using [usbtreeview](https://www.uwe-sieber.de/usbtreeview_e.html) to check the config descriptor before / after a rebuild changing the string, it works as expected, i.e. Win isn't shadowing the strings and usbtreeview returns the correct strings.
This would seem like a pretty convenient way of doing this. Thoughts ? @HubertD ?

fenugrec, 2021-02-16:
@normaldotcom , @tuna-f1sh  , you guys also use this and/or have forks, can you think of any reason not to do this by adding a config descriptor as I described ?

This is what `iConfiguration` it looks like with lsusb on linux : 

```
Bus 005 Device 003: ID 1d50:606f OpenMoko, Inc. cannette gs_usb
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
 ........
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0032
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 ec69f1f8-pr62
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              150mA
    Interface Descriptor:
....
```

tuna-f1sh, 2021-02-17:
I can't think of a reason not to, no. Sounds like a useful feature to me.

How are you injecting it to the source? Using a `cmake` function?

fritz-net, 2021-02-17:
Have you looked into `git describe`? If there is space in the description string i would probably use this since it can generate a version number based on tags with an additional git has so there is also a humane readable indication for the current version.

I have used `git describe --tags --match "Version-[0-9].[0-9].[0-9]" --abbrev=6` in the past:
[image]
The last Tag in the example is 474 commit away and is named "Version-3.0.0"

fenugrec, 2021-02-17:
> How are you injecting it to the source? Using a `cmake` function?

TBD. For testing I was just updating it manually; in another project I use some cmake magic I lifted somewhere:

https://github.com/fenugrec/nisprog/blob/master/cmake/gitversion.cmake

I'd probably start with that and see if I need to address the edge-case of someone compiling from a tarball without 'git' available.

fenugrec, 2021-04-12:
I have a PR something usable ready for testing/review, https://github.com/candle-usb/candleLight_fw/pull/71

fenugrec, 2021-04-25:
Closed via #71

closed 2021-04-25 by fenugrec