candle-usb/candleLight_fw #30: CANAlyze support

brandonros, 2019-11-08:
Does this work with https://kkuchera.github.io/canalyze/?

fenugrec, 2019-11-08:
Maybe, but only if it uses the same pin assignments as other F042-based hardware that is already supported here. You would have to check the schematics, and verify against
https://github.com/candle-usb/candleLight_fw/blob/master/include/config.h for the LED pins and
https://github.com/candle-usb/candleLight_fw/blob/master/src/can.c#L29  for the CAN pins.

HubertD, 2019-11-08:
CAN is on PB8/PB9 and LEDs are on PB0/PB1.
Looks like canalyze should be compatible with the cantact, which is supported by our firmware.

brandonros, 2019-11-08:
> Looks like canalyze should be compatible with the cantact, which is supported by our firmware.

For super mega clarity, does this mean that if I put CANAlyze in DFU mode, flash candlight-fw, it would be work out of the box with no changes?

HubertD, 2019-11-08:
won't give you a guarantee, but I think so.
use the cantact flavor.

would be glad to hear about the outcome.

brandonros, 2019-11-08:
I will test right now and if it works, I will make a PR to update the README that CANAlyze is supported

Question:

[image]

brandonros, 2019-11-09:
More recent firmware for other people: https://github.com/candle-usb/candleLight_fw/issues/32#issuecomment-552128111

fenugrec, 2019-11-11:
Yes, we could / should tag a new release. Starting a new issue for that.

closed 2019-11-11 by fenugrec

brandonros, 2019-11-11:
> CAN is on PB8/PB9 and LEDs are on PB0/PB1.
> Looks like canalyze should be compatible with the cantact, which is supported by our firmware.

@fenugrec can you double check I got the LED pins right first?