candle-usb/candleLight_fw #32: /usr/bin/objcopy: Unable to recognise the format of the input file `candleLight_fw'

brandonros, 2019-11-09:
```
 CC='/usr/bin/arm-none-eabi-gcc' CFLAGS="--specs=nosys.specs" CMAKE_OBJCOPY="/usr/lib/arm-none-eabi/bin/objcopy" cmake .
```

It's very odd how CMAKE_OBJCOPY is not successfully being set and /usr/bin/objcopy is being used instead.

brandonros, 2019-11-09:
I had to manually edit `/usr/bin/objcopy` to `/usr/lib/arm-none-eabi/bin/objcopy` to get the result.

```
-rwxr-xr-x  1 brandon brandon  16068 Nov  9 13:48 candleLight_fw.bin
```

```
$ md5sum candleLight_fw.bin 
15923016d0d41f28a9214fe4b5f2f81f  candleLight_fw.bin
```

git commit 9e4938492b56b48b487fc3baf199550e3e62d41e

brandonros, 2019-11-09:
[Uploading candleLight_fw_9e49384.bin.zip…]()

brandonros, 2019-11-09:
[candleLight_fw_9e49384.bin.zip](https://github.com/candle-usb/candleLight_fw/files/3827614/candleLight_fw_9e49384.bin.zip)

fenugrec, 2019-11-09:
> I had to manually edit `/usr/bin/objcopy` to `/usr/lib/arm-none-eabi/bin/objcopy`

whoa, whoa, back a few steps : did you specify the toolchain file when first running cmake ?

brandonros, 2019-11-09:
> > I had to manually edit `/usr/bin/objcopy` to `/usr/lib/arm-none-eabi/bin/objcopy`
> 
> whoa, whoa, back a few steps : did you specify the toolchain file when first running cmake ?

Where is that documented? Apologies

brandonros, 2019-11-09:
[cantact_fw_9e49384.bin.zip](https://github.com/candle-usb/candleLight_fw/files/3827626/cantact_fw_9e49384.bin.zip)

fenugrec, 2019-11-11:
If you're using cmake-gui, delete cache, start over and "select compiler toolchain", point to the included file  cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake

brandonros, 2019-11-11:
`cmake . -DCMAKE_TOOLCHAIN_FILE=cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake `

closed 2019-11-11 by brandonros