candle-usb/candleLight_fw #34: [RFC] tag new release

fenugrec, 2019-11-11:
As was noted in another issue, there hasn't been a tagged release in a long time, which is annoying for anyone that doesn't want to install gcc-arm and compile themselves.

Do we tag one right now, and also later after the st_libs update; or wait to merge the update before and tag at that time ?

HubertD, 2019-11-13:
we should definitely put a version 2 out there very soon.
but I would like to include the libs update since it is a rather large change and hopefully a good basis for maintaining a v2.0

fenugrec, 2019-11-13:
Ok. How about a 1.1 now and 2.0 with new libs ?

fenugrec, 2022-04-19:
Just tagged v2.0 today. I decided with new libs, lots of patches, it qualified for a different major number !

closed 2022-04-19 by fenugrec