candle-usb/candleLight_fw #38: Where to find documentation of gs_usb iterface?

Jochen0x90h, 2019-11-28:
Hi!
I'd like to use the firmware with libusb e.g. from MacOS. Is there some documentation of the gs_usb interface that you implemented? With libusb I can directly send requests to the usb device so all I need is to set the data rate (e.g. to 500kHz for OBDII) and then send and receive data packets.

Thanks!

vondraussen, 2019-11-30:
@Jochen0x90h I think the Kernel module source can be a starting point:
[https://github.com/torvalds/linux/blob/master/drivers/net/can/usb/gs_usb.c](https://github.com/torvalds/linux/blob/master/drivers/net/can/usb/gs_usb.c)

daniel-santos, 2019-12-01:
@Jochen0x90h Yes, I'm afraid that's about it.  In general, there is [can.rst](https://www.kernel.org/doc/Documentation/networking/can.rst) / [can.txt](https://www.kernel.org/doc/Documentation/networking/can.txt) in the kernel Documentation, but that's for the higher-level access to CAN devices in Linux.

To better understand gs_usb.c, you'll need to read it and also understand the various sub-systems it's implementing devices in.  I also know that Ethan has some info on OSX [here](https://canable.io/getting-started.html).  Sorry I don't have much more for you as I don't use Mac.

daniel-santos, 2019-12-01:
@Jochen0x90h I think I answered before fully reading yourn question. :)  Your answer is mostly in the packed structs.  `gs_device_config` sends the capabilities of the device to the host, and the host keeps `GS_MAX_RX_URBS` (30) bulk IN URB requests outstanding at a time.

I'm planning on submitting a proposal to extend this driver to support CAN FD and also to make it more efficient for CAN and CAN FD.  This discussion will take place on LKML and probably linux-usb and linux-can.  When i finally get around to it, I'll announce it here.

Jochen0x90h, 2019-12-02:
@daniel-santos Thanks. I have to start experimenting with it, but if you know how to fill out gs_device_bittiming to get 500kHz (for OBDII port) this would be a good starting point

Jochen0x90h, 2019-12-06:
I'm now able to initialize the device and set the bit timing, but simply sending and receiving frames doesn't do the trick yet.

Here's my code: https://github.com/Jochen0x90h/candlelight-libusb/blob/master/src/main.cpp

I receive garbage in the frame but I would expect that I receive nothing if nothing is connected to the can bus. When I connect it to the OBD-II Port of a car then I still receive garbage. Is there anything I miss?

Jochen0x90h, 2020-01-10:
@vondraussen Now I got it to work. These problems I still have with a CANable 0.2 device:

1. In loopback mode, I get two responses with different timestamps while I would expect only one

2. Sometimes I can read some or many frames after disable/enable and before a frame was written. Termination is set. this seems to be an issue with the buffer management, maybe the queues are not cleared when the can gets disabled.

3. Sometimes I don't get a reply to a real OBD-II request to a car on the first try, but on the second.

fenugrec, 2022-08-27:
Closing this due to inactivity, and because there's not much documentation beyond the linux driver sourcecode.

closed 2022-08-27 by fenugrec