candle-usb/candleLight_fw #40: candleLight for BluePill

fritz-net, 2019-12-19:
Hello guys, i tried porting this project to stm32f103/bluePill with no success since I can't find a compatible HAL lib.

Do you have any ideas where I can find an HAL lib which works with this project and stm32f1xx cores? Or are there any plans to support these cores? They seem to be pin compatible.

fenugrec, 2019-12-19:
Hi,
the F103 is an old part, and it doesn't support simultaneous USB + CAN (see the ref manual). The much less common f105/f107 parts do IIRC but there's no reason to use them with newer, cheaper parts available.

closed 2019-12-19 by fenugrec

fritz-net, 2019-12-24:
I didn't know that they can't do this. Thank you for clarifying this.

[image]

brandonros, 2020-05-03:
https://github.com/NerdyProjects/usb2can_stm32f103

http://randomport.com/pages/usb-to-can.html

why isn't this chip supported again? it seems possible

fritz-net, 2020-05-03:
cause in both projects USB is not handled by the microcontroller. Instead they use an external usb chip which has a lot of downsides. (maybe there is also a bottleneck)

fenugrec, 2020-05-03:
> why isn't this chip supported again? it seems possible

Uhh, I answered two posts earlier, fritz-net posted the exact reference in the RM, and that second link you posted says the same, 

> Although the STM32F103 MCU supports USB natively, it unfortunately cannot use USB and CAN at the same time. The USB to serial conversion is therefore done by FT230XS IC.

I'm not sure what else to add.

brandonros, 2020-05-03:
> it unfortunately cannot use USB and CAN at the same time

that’s extremely mislead because tons of USB CAN devices using STM32F103 exist and the only way they function is because they can send + receive CAN frames over USB

i get what you are saying that it is handled by a USB -> UART chip or whatever, but it was just worded weird

fritz-net, 2020-05-03:
I think adding UART stuff to candlelight would add to much noise to the code base and would not be "single responsibility". You also linked some firmware doing exactly this (can -> stm32f013 -> uart -> usb)