candle-usb/candleLight_fw #42: What do these error frames mean?

brandonros, 2020-01-19:
I get them back when trying to send a message to a separate CAN controller. I've tried flipping the termination pin and flipping CAN high/CAN low: no avail.

```
ffffffff0c000020080000000000080000000000
ffffffff0c00002008000000000c080000000000
ffffffff0c000020080000000030080000000000
ffffffff4c00002008000000000c080000000000
ffffffff4c000020080000000030080000000000
```

They are error frames, I just don't know what they mean.

fritz-net, 2020-04-10:
The format u gave is horrible to read. Just read this definitions here:
https://github.com/candle-usb/candleLight_fw/blob/9003f5ec1d27dbb5bdae67d74d53ba57dc1616ab/include/gs_usb.h#L63

fenugrec, 2022-08-27:
Closing for inactivity and lack of information.

closed 2022-08-27 by fenugrec