candle-usb/candleLight_fw #43: LEDs on CANable

normaldotcom, 2020-02-24:
The LEDs on the CANable with candlelight can be a bit confusing (no LEDs illuminated on powerup)--I've updated it in my forked repo (which is way behind) but I would like to get some changes in the upstream repo if possible.

I'll pull a PR together for the LED state changes, but I would also like to add blinking of both LEDs on powerup. Is there a preferred place I should add this to the CANable-specific code? This could also apply to other boards if that would make sense.

FYI, for the LEDs I simply swapped the polarity for the Green LED.
 - Interface down: Blue LED is "on", Green LED is "off" (indicates that the device is powered up)
 - Interface up: Blue LED is "off", Green LED is "on" (indicates that the interface is up)

Thanks!

fenugrec, 2020-02-24:
Hi,
did you see PR #37 ? I'm also not a huge fan of the current behaviour but it does have some logic to it.

> single blink on power-up 

Ok, didn't need that with my hardware (has a dedicated power LED) but I can imagine no argument against this. It could probably go somewhere in main.c near the call to led_set_mode( ) ? (common to all boards)

> interface down/up

heh, we now have 3 different ideas on what the LEDs should show ! Personally I prefer tx vs rx activity, Hubert wanted it to behave like an ethernet jack, and you have even something else in mind. I'm not sure how to reconcile all this without spiraling in an #ifdef mess... maybe something like I described in that PR , with an added "heartbeat" blink when power is on but interface is down ?

normaldotcom, 2020-02-25:
Didn't see that pull request, glad someone has been thinking about it! I think I can make a clean PR adding blinking of LEDs on powerup, and then just change the polarity of the green LED in the CANable-specific section of config.h. Should avoid a big #ifdef mess, definitely don't want to make things messy.

fenugrec, 2020-10-31:
As you probably saw I merged your LED PR. If behaviour is OK on your board types now I'll close this. Thanks !

fenugrec, 2020-12-06:
Feel free to reopen if any further issues.

closed 2020-12-06 by fenugrec