candle-usb/candleLight_fw #44: ARM Cortex-M3 STM32F103C8T6 STM32 Minimum System Development Board With Mini USB

brandonros, 2020-04-07:
https://www.tindie.com/products/mmm999/arm-cortex-m3-stm32f103c8t6-stm32-minimum-system-development-board-with-mini-usb/

Would this device work? If not, why not? I am down to buy one and work on a PR to get it to compile for this device.

fenugrec, 2020-04-07:
Never. F103 cannot do simultaneous USB + CAN. It's subtly mentioned once or twice in the RM.

closed 2020-04-07 by brandonros

brandonros, 2020-04-08:
for anybody else who lands on this but is unfamiliar with stm32 family:

```
STM32 H7
STM32 F7
STM32 F4
STM32 F3
STM32 F2
STM32 F1
STM32 F0
STM32 G0
STM32 L5
STM32 L4
STM32 L1
STM32 L0
STM32 W
STM32 J
```

this repo talks about stm32f

is it *just* stm32f0, or do you support these as well:

```
STM32 F7
STM32 F4
STM32 F3
STM32 F2
STM32 F1
STM32 F0
```

pretty much all of the supported built ready to order devices in the README are out of stock/discontinued so I'm trying to find a modern day alternative

brandonros, 2020-04-08:
looks like only the `stm32f0x2` line is supported. no other `f0x` line can do both CAN + USB

options:

```
STM32F042G6
STM32F042F6
STM32F042G4
STM32F042F4
STM32F042T6
STM32F042K6
STM32F042K4
STM32F042C6 (canable)
STM32F042C4
STM32F072C8
STM32F072CB
STM32F072RB
STM32F072R8
STM32F072VB
STM32F072V8
```