candle-usb/candleLight_fw #47: objcopy should use arm-linux-eabi-objcopy

romain145, 2020-04-11:
Compilation fails on Ubuntu 19.10 with the following error :

```
/usr/bin/objcopy: Unable to recognise the format of the input file `canable_fw'
make[3]: *** [CMakeFiles/canable_fw.dir/build.make:320: canable_fw] Error 1
make[3]: *** Deleting file 'canable_fw'
make[2]: *** [CMakeFiles/Makefile2:150: CMakeFiles/canable_fw.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:162: CMakeFiles/canable_fw.dir/rule] Error 2
make: *** [Makefile:144: canable_fw] Error 2
```

In CMakeFiles/canable_fw.dir/build.make:320 the command should use the cross-compiler provided objcopy as follows:
`arm-none-eabi-objcopy -O binary canable_fw canable_fw.bin`

Then compilation succeeds:

```
romain@romain-PC:~/candleLight_fw/build$ make canable_fw
[ 58%] Built target STM32_HAL_STM32F042x6
[ 67%] Built target STM32_USB_Device_Library_STM32F042x6
[ 70%] Linking C executable canable_fw
   text	   data	    bss	    dec	    hex	filename
  15407	    536	   2012	  17955	   4623	canable_fw
[100%] Built target canable_f
```

It might be fixed in the CMakeLists.txt but I'm not CMake-literate...

fenugrec, 2020-04-11:
Hi,
the issue title is a bit confusing since "arm-linux-eabi-objcopy" is definitely not the right toolchain for this. Do you have a gcc-arm-linux-* package installed or only gcc-arm-none-eabi ?

fenugrec, 2020-04-11:
can you empty your build/ dir completely, re-run cmake and pastebin CMakeCache.txt somewhere ?

romain145, 2020-04-11:
Yes it is because that's the one used to compile.

See CMake:

```
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)

set(CMAKE_C_COMPILER "${TOOLCHAIN_BIN_DIR}/arm-none-eabi-gcc" CACHE INTERNAL "")
set(CMAKE_CXX_COMPILER "${TOOLCHAIN_BIN_DIR}/arm-none-eabi-g++" CACHE INTERNAL "")
set(CMAKE_EXE_LINKER_FLAGS "" CACHE INTERNAL "")
```

romain145, 2020-04-11:
CMakeCache.txt file: https://pastebin.com/3rFVhr4W

romain145, 2020-04-11:
Ok, for some reason this is compiling now...
Closing this .
You can reject the PR. Tks.

closed 2020-04-11 by romain145

fenugrec, 2020-04-11:
Ah, I thought this was weird. Just note how your cmakecache has `CMAKE_OBJCOPY:FILEPATH=/usr/bin/arm-none-eabi-objcopy` properly set.