candle-usb/candleLight_fw #50: proper way to use echo_id

brandonros, 2020-05-22:
```javascript
async sendCanFrame(arbitrationId, data) {
    const frame = Buffer.alloc(0x14)
    frame.writeUInt32LE(0xFFFFFFFF, 0x00) // echo_id
    frame.writeUInt32LE(arbitrationId, 0x04) // can_id
    frame.writeUInt8(0x08, 0x08) // can_dlc
    frame.writeUInt8(0x00, 0x09) // channel
    frame.writeUInt8(0x00, 0x0A) // flags
    frame.writeUInt8(0x00, 0x0B) // reserved
    frame.writeUInt8(data[0], 0x0C)
    frame.writeUInt8(data[1], 0x0D)
    frame.writeUInt8(data[2], 0x0E)
    frame.writeUInt8(data[3], 0x0F)
    frame.writeUInt8(data[4], 0x10)
    frame.writeUInt8(data[5], 0x11)
    frame.writeUInt8(data[6], 0x12)
    frame.writeUInt8(data[7], 0x13)
    return this.device.transferOut(this.outEndpoint.endpointNumber, frame)
  }
```

When I write a CAN frame over USB, it gets echoed back to me. Is there a way to *not* do this?

fritz-net, 2020-05-22:
as i remember it is not echoed, it gives the frame on the bus with proper time (which is a very helpful feature - I think youre writting a driver impl for js as i remember, i would keep this feature). The easiest solution would be handling this in your wrapper.

brandonros, 2020-05-23:
it is echoed in the sense that if i TX a packet, i RX it right back. is there a way to filter RX packets based solely on what the device actually receives? i’m doing UDS flashing and the response times/bandwidth are being affected by the fact that whatever I TX, I RX

Regards,

Brandon

fritz-net, 2020-05-23:
as i know there is no possibility in the gs_usb  driver to disable this. Are you sure the bottleneck is USB? If your dive does not replay with the exact same frame you could filter your tx in your wrapper/driver (save last TX -> compare (with timeout) -> ignore)

brandonros, 2020-08-05:
Am I supposed to reset and start *2* channels? 1 for read, 1 for write?