candle-usb/candleLight_fw #53: Firmware Error Handling

GBert, 2020-08-26:
There seems to be a problem with the firmware after an error on the CAN Bus occured:
```
 (2020-08-26 10:16:07.727451)  can0  RX - -  0030A301   [0]                            ''
 (2020-08-26 10:16:07.727922)  can0  RX - -  00316F13   [8]  43 53 CE C0 03 51 00 00   'CS...Q..'
 (2020-08-26 10:16:09.812794)  can0  RX - -  2000000C   [8]  00 00 04 00 00 00 00 00   ERRORFRAME
	controller-problem{}
	protocol-violation{{bit-stuffing-error}{}}
```
which leads to any valid CAN frame afterwards will be attended by  
```
 (2020-08-26 10:16:09.813001)  can0  RX - -  2000000C   [8]  00 00 02 00 00 00 00 00   ERRORFRAME
	controller-problem{}
	protocol-violation{{frame-format-error}{}}
```

and later attended by
```
 (2020-08-26 10:17:30.235544)  can0  RX - -  20000004   [8]  00 0C 00 00 00 00 00 00   ERRORFRAME
	controller-problem{rx-error-warning,tx-error-warning}
``` 
or
```
(2020-08-26 10:17:49.675803)  can0  RX - -  20000004   [8]  00 00 00 00 00 00 00 00   ERRORFRAME
	controller-problem{}
```
Some additional frames later these error frames disappear and the CAN adapter seems to work as expected.
Checked with usbmon that the wrong error frames come from the USB device.

This error is reproducible.

Environment:
------------------
actual candlelight_fw/cantact firmware on [Electrodragon cantact clone](https://www.electrodragon.com/product/can-usb-debugger-board/) 
Ubuntu Linux 20.04 Linux 5.4.0-42-generic with 
`sudo ip link set can0 up type can bitrate 250000 restart-ms 100`

brian-brt, 2020-12-02:
What behavior are you expecting? No error frames, error frames in a different order, error frames with different contents, something else?

I've been working on the contents of these error frames over in #59, for reference.

fenugrec, 2022-08-27:
This was probably fixed by pr #61  .

closed 2022-08-27 by fenugrec