candle-usb/candleLight_fw #65: STM32F042x6 series cannot do both USB+CAN+send/receive

brainstorm, 2020-12-29:
**TL;DR: Does anybody know a solution to get USB+CAN+dump+send working on the `STM32F042G6` with UFQFPN28 packaging ? I'm all ears :)**

On the README.md there are several  it reads:

> Of important note is that the common STM32F103 will NOT work with this firmware because its hardware cannot use both USB and CAN simultaneously.

Isn't it true for the STM32F042x6 series too? According to [this source](https://github.com/majbthrd/CANsniffer#hardware-tradeoffs):

> The designers of the STMicro STM32F0xx family of parts made the unfortunate decision to assign USB and CAN peripheral functionality to the same pins.
> 
> There is an additional pin assignment option, but it is for CAN_RX only. As a result, this implementation is forced to be receive-only (e.g. a sniffer).

I have [a small UFQFPN28 4x4 mm package for my `STM32F042G6`](https://github.com/brainstorm/bbtrackball-rs) on a dev board that has 28 pins:

https://github.com/candle-usb/candleLight_fw/blob/master/libs/STM32_HAL/include/stm32f0xx/stm32f0xx_hal.h#L96

[image]

And I had to do the following minor changes on the firmware to get the USB working/enumerating:

```diff
diff --git a/src/main.c b/src/main.c
index 47b63b5..22c28f8 100644
--- a/src/main.c
+++ b/src/main.c
@@ -215,6 +215,8 @@ void SystemClock_Config(void)

        HAL_SYSTICK_CLKSourceConfig(SYSTICK_CLKSOURCE_HCLK);

+       __HAL_REMAP_PIN_ENABLE(HAL_REMAP_PA11_PA12);
+
        /* SysTick_IRQn interrupt configuration */
        HAL_NVIC_SetPriority(SysTick_IRQn, 0, 0);
 }
diff --git a/src/can.c b/src/can.c
index 80c0a54..235ea14 100644
--- a/src/can.c
+++ b/src/can.c
@@ -32,7 +32,7 @@ void can_init(can_data_t *hcan, CAN_TypeDef *instance)
        __HAL_RCC_CAN1_CLK_ENABLE();

        GPIO_InitTypeDef itd;
-       itd.Pin = GPIO_PIN_8|GPIO_PIN_9;
+       itd.Pin = GPIO_PIN_3|GPIO_PIN_4;
        itd.Mode = GPIO_MODE_AF_PP;
        itd.Pull = GPIO_NOPULL;
        itd.Speed = GPIO_SPEED_FREQ_HIGH;
```

But AFAICT it's game over for this IC/package if I want to have a functional send/receive USB+CAN board, right? Perhaps it should be stated in the README alongside the `STM32F103` remark?

fenugrec, 2020-12-29:
> Isn't it true for the STM32F042x6 series too?

No. I have boards with a F042C6T6 (lqfp48) working, both TX+RX. The source you cite is only right for smaller packages , 28 and 32-pin I believe.

> But AFAICT it's game over for this IC/package if I want to have a functional send/receive USB+CAN board, right?

Correct.

> Perhaps it should be stated in the README alongside the `STM32F103` remark?

That remark was mainly because we were receiving many questions about the F103, and that IC's limitation (shared buffer mem) is really not obvious when skimming over the datasheet IIRC it's only mentioned in 1 place in the RM.
I'd argue the F042 case is more evident : just looking at the pin assignments shows that only the 48-pin variant will work.

fenugrec, 2020-12-29:
I added a comment in 68df7d5 .

closed 2020-12-29 by fenugrec