candle-usb/candleLight_fw #70: CANable Pro & candleLight_fw: doesn't work anymore

Ciusss89, 2021-03-19:
I used this CAN adapter (CANable PRO) with candleLight fw to develop my thesis project. It has worked well.

I tested the system today but seems that the CAN adapter has a problem with last Linux/firmware (or both):
1. I'm running fedora 33 with `Linux vaio 5.10.19-200.fc33.x86_64 #1 SMP Fri Feb 26 16:21:30 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
`
2. Looking into kernel commit I noted this -> [can: gs_usb: fix endianess problem with candleLight firmware](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=4ba1cb39fce4464151517a37ce0ac0a1a3f580d6)
3.  Canble pro update page reports new firmware availble:
4. I tested both (`a8a0757` and `68df7d5` 1/3/21 BETA) and doesn't work.

---------------------------------------------------------------------------------------------------
1. 
```
~ $ dmesg  | tail -n 7
[59704.827955] usb 3-2: new full-speed USB device number 15 using xhci_hcd
[59704.955316] usb 3-2: New USB device found, idVendor=1d50, idProduct=606f, bcdDevice= 0.00
[59704.955323] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[59704.955326] usb 3-2: Product: canable gs_usb
[59704.955328] usb 3-2: Manufacturer: canable.io
[59704.955330] usb 3-2: SerialNumber: 00020009524E430220373533
[59704.957981] gs_usb 3-2:1.0: Configuring for 1 interfaces
```

[image]

fenugrec, 2021-03-19:
I observe two things here.

1- I can have similar behavior if there's just one CAN device (the candlelight) on the bus, i.e. no other device to generate the ACKs. Why does candump see 3 echo frames ? There should be none.

2- "no buffer space available" : this is normal, and *should* rectify itself if you activate a second CAN node on the bus.

> works fine with slcand
Are you saying a single slcan device shows echo'd frames even with no other node to generate the ACK ? That doesn't sound right.

fenugrec, 2021-04-25:
@Ciusss89  anything new ? Have you tested with at least 2 devices on the bus as I mentioned ?

Ciusss89, 2021-04-27:
I miss your replay. I'm apologize for the delay! I will do a new test this weekend!

fenugrec, 2022-01-06:
Closing this due to no info. Feel free to reopen

closed 2022-01-06 by fenugrec