candle-usb/candleLight_fw #72: STM32F105 compatibility

segaura, 2021-04-20:
I am considering the possiblity of putting candleLight on a board with STM32F105RBT6 and two tranceivers and, knowing of the incompatibilty with STM32F103, I am wondering if the same kind of issues stand for STM32F015.

That board is the maybe-some-of-you-already-know-about "18 in 1 canbus filter" which actually have no default breakout usb so feasibility of the whole operation is not obvious, even if the firmware and the silicon are compatible.

segaura, 2021-04-20:
Regarding OTG, it seems that STM uses the term as smartphone manufacturers do, [it can be used both as host and as device](https://www.st.com/resource/en/user_manual/cd00289278-stm32f105xx-stm32f107xx-stm32f2xx-and-stm32f4xx-usb-onthego-host-and-device-library-stmicroelectronics.pdf).

At least for what I am able to understand from this document (close to nothing :-))

[image]

fenugrec, 2021-04-25:
Last I checked the F105 was quite expensive vs its capabilities, compared to the F042/F072 we currently support... So what advantage would it have in your application ? the USB driver is also very different (OTG as you pointed out), that means more code to test...

Sure, it would be possible to split out the entire USB layer into different backends and support even more different devices, but I don't think it would be worth the effort.

segaura, 2021-04-26:
I agree with you, my investigation started when I've seen some very cheap ready-made modules selling on asiatic platforms and related to not-such-ethycal activities like mileage correction or similar.
They are usually referred to as "18 in 1 universal can filter", they have an STM32F105 and two tranceivers and because of their "close-to-ten-dollars" price range I asked myself if it could be reasonable to rewrite their firmware and make them general purpose, specifically for those use cases where two bus are an added value, like in automotive application (e.g forscan).
It is correct to give up because:
- chip is not already supported
- candlelight manages only one can channel (as far as I understood)
- STM32 usb pins on these boards are not easily accessible
- thinking that these silicons are "original" and that issues will not arise using them is against all odds

Thank you anyway for your time.

fritz-net, 2021-04-26:
@segaura I thought this OBD devices you mentioned are connected via USB to the computer -> which would makle there USB pins accessible, even with a nice connector. Programming pins could be harder. Do you have a hardware sample?

segaura, 2021-04-26:
They are not exactly OBD, have a look at this sample picture

[image]

they sit just before the odometer and for each frame with data of new distance travelled they forward an identical frame with zeroed data, instead.

In other words they are autonomous and placed usually in a hidden location, with no USB necessity.
When I read about a guy that reversed engineered them and dumped the firmware, I asked myself if they can be repurposed, but that seems not an easy task.

fenugrec, 2021-04-26:
@segaura , thanks for the extra info. Yeah, with the USB pins not routed, it's going to be some work.

I'm not sure what would be missing to have candlelight support > 1 CAN interfaces; I know there are provisions in the code for this both in the firmware and in the gs_usb kernel driver, but no idea of the status and completeness.

Currently it's probably easier and cheaper to just have two candle devices if you need 2 separate buses : )

fritz-net, 2021-04-26:
@segaura If they have no USB I don't think they make a lot of sense since you have to do a lot of additional work ...

I manufactured/designed my Hardware myself and mine has an USB connector and was costing about 20$ ... If you spend a little more you can get isolation and so on.
The OBD devices would have been nice since they have a proper case (and sometimes a D-SUB connector)

segaura, 2021-04-26:
really cool, and I dream af being able to do it also, but in term of effort we are talking of a one or two order of magnitute difference, IMHO ...half of my brain is full of JLCPCB and PCBWay commercials I am able to driblle on youtube :-) so someday I will try anyway

fenugrec, 2021-04-26:
In any case, I'll close this for the moment :

- F105 currently not supported, and not planned
- F105 could theoretically work (I believe it can mux both CAN and USB to different pins even in the 64-pin package), but would need a different USB stack due to the OTG periphal, and possibly upstream changes to gs_usb driver for multi-channel
- the cheap F105 board with 2 CAN transceivers shown above doesn't have USB routed and would be a pain to reuse

closed 2021-04-26 by fenugrec