candle-usb/candleLight_fw #73: How to manually set baud rate?

brandonros, 2021-05-04:
Is it always automatically calculated?

brandonros, 2021-05-04:
```
static int gs_usb_set_bittiming(struct net_device *netdev)
{
	struct gs_can *dev = netdev_priv(netdev);
	struct can_bittiming *bt = &dev->can.bittiming;
	struct usb_interface *intf = dev->iface;
	int rc;
	struct gs_device_bittiming *dbt;

	dbt = kmalloc(sizeof(*dbt), GFP_KERNEL);
	if (!dbt)
		return -ENOMEM;

	dbt->prop_seg = cpu_to_le32(bt->prop_seg);
	dbt->phase_seg1 = cpu_to_le32(bt->phase_seg1);
	dbt->phase_seg2 = cpu_to_le32(bt->phase_seg2);
	dbt->sjw = cpu_to_le32(bt->sjw);
	dbt->brp = cpu_to_le32(bt->brp);

	/* request bit timings */
	rc = usb_control_msg(interface_to_usbdev(intf),
			     usb_sndctrlpipe(interface_to_usbdev(intf), 0),
			     GS_USB_BREQ_BITTIMING,
			     USB_DIR_OUT | USB_TYPE_VENDOR | USB_RECIP_INTERFACE,
			     dev->channel,
			     0,
			     dbt,
			     sizeof(*dbt),
			     1000);

	kfree(dbt);

	if (rc < 0)
		dev_err(netdev->dev.parent, "Couldn't set bittimings (err=%d)",
			rc);

	return (rc > 0) ? 0 : rc;
}
```

looks like I need to know what 

```
         dbt->prop_seg = cpu_to_le32(bt->prop_seg);
	dbt->phase_seg1 = cpu_to_le32(bt->phase_seg1);
	dbt->phase_seg2 = cpu_to_le32(bt->phase_seg2);
	dbt->sjw = cpu_to_le32(bt->sjw);
	dbt->brp = cpu_to_le32(bt->brp);
```

these values should look like at 500k and 1mb if I understand correctly

marckleinebudde, 2022-01-08:
Hey @brandonros,

this is the git repo of the candleLight firmware, not the Linux driver, which you are referring to. But I can help you anyways:

In Linux you set the bitrate with the `ip link` command:

```
sudo ip link set can0 type can bitrate 125000
```

Here the kernel calculates the from the given bitrate (here 125 kbit/s) the bit timing parameters (`prop-seg`, `phase_seg1`, ....) and the driver pushes them over USB into the device.

But you can define these bit timing parameter yourself, e.g.:

```
sudo ip link set can0 type can tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
```

Look here for more details: https://elixir.bootlin.com/linux/v5.15/source/Documentation/networking/can.rst#L1300

fenugrec, 2022-04-17:
I think this can be closed.

closed 2022-04-17 by fenugrec