candle-usb/candleLight_fw #74: Contract help wanted to port Candlelight to Artery AT32F415

lsoltero, 2021-07-10:
Hello All,

as you know by now STM MCUs are in very short supply and very expensive these days if you can even find them.  We have seen prices go from $1.80 -> $10 -> $29 two weeks ago and still rising.

We have been looking a replacing the STM MCUs with the Artery AT32F415.  The Artery MCU as very similar in architecture to the STM units.   We have been able to load the candlelight firmware onto these units but it fails to run and enumerate.

Here is a link to the Artery MCU programming guide.

https://www.arterytek.com/download/RM_AT32F415_EN_V1.03.pdf

Wondering if anyone in the community would be interested in porting Cadelight to this new platform.  If so please contact me with contract terms and availabilty.

Thanks,

--luis

fenugrec, 2021-07-10:
p. 487
> USB and CAN share a dedicated 512-byte SRAM memory for data transmission and reception, so they cannot be used concurrently (the shared SRAM is accessed through CAN and USB exclusively). USB and CAN can be used in the same application but not at the same time.

lsoltero, 2021-07-10:
I have confirmed with Artery tech support that this is no longer the case and that the manual is old and out of date.

Happy to share my correspondence with Artery on this subject.

Please be advised that the CAN and USB can run concurrently with the current AT32F415 MCUs

lsoltero, 2021-07-10:
you may find this helpful and i quote.

```
Please check below more details.

Artery and ST, they use the same USB core, so the driver is the same. DFU function is the same too. You can check below link sample codes and documents.

AT32_IAP_using_the_USB_HID Sample code and documents

 

https://www.arterychip.com/download/AN0007_AT32_IAP_using_the_USB_HID_EN_V1.0.2.pdf

https://www.arterychip.com/download/AN0007_AT32_IAP_using_the_USB_HID_EN_V1.0.2.zip
```

and some correspondence that is relevant to this...

```
so.... the engineer at AsiaRF spent some time with ARTERY on the phone today... As it turns out the documents they have
published on their web are for version 1.03... they are shipping v 1.3 which is much newer and...

NO LONGER LIMITS THE USE OF USB and CAN AT THE SAME TIME!

the attached data sheet specifically states that the USB Host controller now has 1280 bytes of dedicated SRAM.

so it seems that they have addressed the issue.  The factory has stated that there are no issues running both interfaces
at the same time.

here is a link to additional technical resources.  I have been told to look specifically at

https://www.dropbox.com/sh/ysupmty6mk7yrlt/AABEOKUU4xEUrAHN6Y3d2-5Za?dl=0

and I quote...

IC company confirmed, USB and CAN can work in the meantime. but
sorry there is no sample codes yet. please refer BSP's VCP_Loopback add on CAN function, sample code directory is below:
..\AT32F4xx_StdPeriph_Lib_V1.3.0\Project\AT_START_F415\Examples\OTG_Device\VCP_Loopback

you will find samples in the zip files in the linkabove.
```

fenugrec, 2021-07-10:
Thanks for the info. I don't have time to work on this but wanted to clarify that point since it's the more or less well-known limitation of the F103

lsoltero, 2021-07-10:
Yes... this is a VERY important issue and one that we were concerned about early on... So... spent some time researching it.

here is the pertinent description of the USB in their latest datasheet.   I quote...

- CAN interface (2.0B active) with dedicated 256 bytes SRAM
- USB2.0 full-speed device/host/OTG controller with dedicated 1280 bytes SRAM, device mode supporting crystal-less

[image]

lsoltero, 2021-07-10:
I am thinking that this should be a pretty straightforward project for someone familiar with the STM development tools and architecture.

vondraussen, 2022-01-08:
> Hello All,
>
> as you know by now STM MCUs are in very short supply and very expensive these days if you can even find them. We have seen prices go from $1.80 -> $10 -> $29 two weeks ago and still [...]

I think I should start selling the parts I still have on stock 🤑