candle-usb/candleLight_fw #75: It seems something wrong with version.h in my build

iDoka, 2021-07-12:
When I do `lsusb -d 1d50:606f -v` I get:
```
...
  iSerial                 3 000500125746430120333932
    iConfiguration          4 fw_9195f08_%as+
...
```

The same value contain in the file `version.h`:
```
$ cat version.h 
#define GIT_HASH "fw_9195f08_%as+"
```

As I correct recognize the file `version.h` is automated generate file and "%as+" in define of GIT_HASH looks strange (it seems substitute  "%as+" don't pass)

Whats is main goal of "%as+" ?
Is it normal if I see in output of  `lsusb` the value `fw_9195f08_%as+`?

fenugrec, 2021-07-12:
> Whats is main goal of "%as+" ?

```
$ man git log
....
               %as
                   author date, short format (YYYY-MM-DD)

```

The '+' is then appended if the source tree is dirty (see cmake/gitversion.cmake)

What version of git are you using ?
What do you get if you run `git log --pretty=format:fw_%h_%as -n 1` from the source dir ?
 `

iDoka, 2021-07-12:
My system might be little bit mature:
```
$ cat /etc/redhat-release 
CentOS Linux release 7.9.2009 (Core)
```
```
$ git --version
git version 1.8.3.1
```
But till now I dont have any issues with git usage (but I'm not advanced git user).

```
$ git log --pretty=format:fw_%h_%as -n 1
fw_85424e3_%as
```
May be settings from ` ~/.gitconfig` override git log? :
```
$ cat ~/.gitconfig
..
[format]
  pretty = oneline
..
```

iDoka, 2021-07-12:
PS: I try to comment strings
```
[format]
  pretty = oneline
```
It does no affect to `git log --pretty=format:fw_%h_%as -n 1`

iDoka, 2021-07-12:
If I exclude `a` from string I get:
```
$ git log --pretty=format:fw_%h_%s -n 1
fw_85424e3_make cantact_fw in README.md
```

iDoka, 2021-07-12:
I did `$ man git log`
It seems I use obsolete git version that doesnt use `%as`:
```
           ·   %ad: author date (format respects --date= option)
           ·   %aD: author date, RFC2822 style
           ·   %ar: author date, relative
           ·   %at: author date, UNIX timestamp
           ·   %ai: author date, ISO 8601 format
```

iDoka, 2021-07-12:
My workaround:

1. Using `%at`
2. Using `%ad` + `--date=short`

```
$ git log --pretty=format:fw_%h_%ad --date=short -n 1
fw_85424e3_2021-07-11
```

closed 2021-07-12 by iDoka