candle-usb/candleLight_fw #82: RFC: consider pulling in ST HAL + CMSIS as submodules instead of local copy

fenugrec, 2022-01-03:
drawback : those repos, even when shallow cloned, are huge. [CubeF0](https://github.com/STMicroelectronics/STM32CubeF0/) is > 300MB,[ HAL alone](https://github.com/STMicroelectronics/stm32f0xx_hal_driver) is almost 200 MB.

Daniel-Trevitz, 2022-10-06:
Tiny USB is using: 
https://github.com/STMicroelectronics/stm32f0xx_hal_driver.git
https://github.com/STMicroelectronics/stm32f4xx_hal_driver.git
https://github.com/STMicroelectronics/cmsis_device_f0.git
https://github.com/STMicroelectronics/cmsis_device_f4.git

When checked out they are currently 40MiB for the cmsis set, and 20MiB for the stm set.
--depth=1 clone only drops 2MB for a total of 58MiB

Our paired-down version of all 4 combined is 6.8MiB, and includes our cmake support.

fenugrec, 2022-11-23:
ah, this could help too :

>$  man gitmodules
> ...
>  submodule.<name>.shallow
>            When set to true, a clone of this submodule will be performed as a shallow clone (with a history depth of
>            1) unless the user explicitly asks for a non-shallow clone.
>