candle-usb/candleLight_fw #91: gs_usb_breq: GS_USB_BREQ_GET_USER_ID and GS_USB_BREQ_SET_USER_ID

ryedwards, 2022-01-30:
I am working on a port of candleLight for the STM32G4 and am stepping through updating the flash_flush() carryover function.  I am slowly able to better understand the interaction between the Linux driver and the firmware.  There are two members of the gs_usb_breq, GS_USB_BREQ_GET_USER_ID and GS_USB_BREQ_SET_USER_ID, that are the reason for having the flash functions.  I do not see these being used in either the latest Linux drivers or in cangaroo.  Am I missing something?

The biggest issue I have is that the current code uses a flash page erase to write back a single word.  Since the .user_data section is only defined as a word a page erase will erase a lot more than just that variable.  Does the current linker reserve a whole page (1K on the STM32F04x)?  

There are routines that read an entire page into RAM, erase flash, modify data then write flash (to emulate EEPROM) but I don't see that being done.

Just curious as to the background of the routines in flash.c and how they are intended to function.

fenugrec, 2022-01-30:
>  current code uses a flash page erase to write back a single word.

Agreed, it's a waste - and it gets worse on bigger devices. The F407 has 128kB pages !

> Does the current linker reserve a whole page 

Yes, see in `ldscripts/ ` , e.g.
```
__NVM_SIZE = 1K;
...
MEMORY
{
  FLASH (rx)  : ORIGIN = 0x08000000, LENGTH = __FLASH_SIZE - __NVM_SIZE
  RAM   (rwx) : ORIGIN = 0x20000000, LENGTH = 6K
  DATA (xrw) : ORIGIN = 0x08000000 + __FLASH_SIZE - __NVM_SIZE, LENGTH = __NVM_SIZE
}

...
```

The only section that should ever end up in DATA, as you found, is `.user_data` . 

I believe this functionality is to emulate behaviour of the original GS_USB hardware. On F042 / F072 losing just 1kB is (usually) not dramatic. I don't think we can fit on the 16kB flash devices anyway ? Haven't checked recently.

If one were to add a new device with the eeprom emulation safely & cleanly disabled I would have no objections to merging. IMO there are better ways to uniquely identify candlelight devices (see #76)

ryedwards, 2022-01-30:
Thanks for the feedback.  The page size on the G47x is 2kB so it's not horrible to reuse the current code.  The only main difference is that it programs a DOUBLEWORD vs. a WORD.

I'm usually good at this stuff but am a little lost on USB fundamentals (this project is a good learning experience for me).  What commands exercise the GS_USB_BREQ_GET_USER_ID and GS_USB_BREQ_SET_USER_ID enumerated values?  What is the purpose of the user_id?

fenugrec, 2022-01-31:
> What commands exercise the GS_USB_BREQ_GET_USER_ID and GS_USB_BREQ_SET_USER_ID enumerated values? 

No idea. AFAIK it's not part of any linux kernel code, so I have to assume it was some user software sending those requests. Maybe some custom configuration sw from the original mfg.. @marckleinebudde  or @HubertD probably know better

marckleinebudde, 2022-01-31:
I haven't seen any code using the `GS_USB_BREQ_GET_USER_ID` or `GS_USB_BREQ_SET_USER_ID`. If you want to test it, I can add it to the lLinux driver.

fenugrec, 2022-01-31:
My own preference would be to drop this part of the code altogether : ))

marckleinebudde, 2022-01-31:
Feel free to remove it, keep the bits, so that they don't get used again for a different purpose.

ryedwards, 2022-01-31:
Of course as soon as I come online today I see this after I got it working last night :)  It needed some tweaking to support multi-channel (was hard coded to only support one channel) but it's there if anyone wants to use it as a reference.  I checked it into my STM32G4 port of the code (https://github.com/ryedwards/CANNEDPI_SW_GSUSB/blob/main/Core/Src/flash.c).

fenugrec, 2022-11-16:
Forgot to close after merging #102

closed 2022-11-16 by fenugrec