candle-usb/candleLight_fw #92: add support for switchable termination

marckleinebudde, 2022-02-15:
Hello

@ryedwards and me are planing to add switchable termination to the gs_usb USB interface. The current idea is to add a new feature bit indicating support, and 2 control messages, one to set and one to get the current termination status. Similar to the `IDENTIFY` feature.

I'll add these to the candleLight firmware repo and send a PR when ready.

fenugrec, 2022-03-01:
Cool. I see mention of something similar in the docs of CANdevstudio, where some drivers support usage such as 
```sudo ip link set can0 type can termination 1```

Is that already in the socketcan layer or other linux drivers, or something entirely driver-specific ?

marckleinebudde, 2022-03-02:
There is complete support form the user space through the kernel into the CAN drivers. The driver has to implement the `do_set_termination` callback and fill out the `termination_const` and `termination_const_cnt` describing the supported resistor values.

marckleinebudde, 2022-09-19:
Let's add the interface bits first. See https://github.com/candle-usb/candleLight_fw/pull/109.

closed 2022-10-06 by marckleinebudde