candle-usb/candleLight_fw #95: Question: Possibility to utilize several CAN interfaces with one controller

bht-patrick-schneider, 2022-05-18:
Hi everyone,
I am at the starting point to dive into this project and just wanted to have a quick question / push in the right direction if there are any objections against what we want to do:
This project is awesome to create a way for extending existing hardware with additional CAN interfaces, but we have use cases where we need more than one additional interface (2 or even 3).
As there are several STM32 controllers that have 2 or 3 CAN interfaces and USB we wanted to look at the possibility to extend this driver to create more than one CAN interface over the gs_usb driver with a single controller and USB connection.
Do you see any objections in terms of technical possibility or performance?
Can you push us in the right direction where to start with our work?

Cheers,

marckleinebudde, 2022-05-18:
Hey Patrick,

first of all the good news, the Linux driver supports 2 interfaces and since v5.18 even 3 interfaces and CAN-FD.

The newest firmware supports STM32F407, e.g. this board: https://stm32-base.org/boards/STM32F407VET6-STM32-F4VE-V2.0.html, don't know how many CAN busses this controller has.

Then there is @ryedwards, who has written a `candleLight`/`gs_usb` compatible firmware for STM32G/STM32H processors supporting IIRC up to 3 channels.

Keep in mind that all channels share the same USB connection, and most controllers only support full speed USB (i.e. 11 Mbit/s). Then I remember there was the limitation of 1KHz USB frames max on some controllers. The current firmware and Linux driver send exactly 1 CAN frame per USB frame, limiting the max amount of CAN frames. But this can be extended.

regards,
Marc

bht-patrick-schneider, 2022-05-18:
Hi Marc,
thanks for the fast reply. That's all the information I'll ever need to get kickstarted. Thanks a lot!

Cheers,
Patrick

closed 2022-05-18 by bht-patrick-schneider