candle-usb/candleLight_fw #96: Unable to install

Hemi940, 2022-06-11:
This is as far as i can get.
pi@raspberrypi:~/build $ cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake
CMake Error at /usr/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:99 (message):
  Could not find toolchain file:
  ../cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

fenugrec, 2022-06-11:
`Could not find toolchain file:
../cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake`

The error is right there... You haven't explained what your source tree looks like, but it's obviously not where you think it is or it's missing files.

Hemi940, 2022-06-11:
When i run this line , cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake
I get the error.
I am unsure what to do

Hemi940, 2022-06-11:
candleLight_gsusb is what im trying to install

fenugrec, 2022-06-11:
... did you even download the source code ?

Actually, maybe you should just download the already-compiled files. https://github.com/candle-usb/candleLight_fw/actions/runs/2478404917

Hemi940, 2022-06-11:
> `Could not find toolchain file: ../cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake`
> 
> The error is right there... You haven't explained what your source tree looks like, but it's obviously not where you think it is or it's missing files.

How should i install the bin file?

Hemi940, 2022-06-11:
How should i install the bin file?

marckleinebudde, 2022-06-11:
The `.bin` file is the software that runs on the USB to CAN adapter. Usually the USB to CAN adapters already have `.bin` file installed. If you want update, follow the instructions in https://github.com/candle-usb/candleLight_fw#flashing.

Hemi940, 2022-06-12:
I'm at a loss
I've downloaded the files and built all the directories, and still get this error.
Do i need the to run BTT U2C 1.1?
I have downloaded the Bin files, but an unsure which one to use.
Any help would be apricated 

pi@raspberrypi:~/build $ cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake
CMake Error at /usr/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:99 (message):
  Could not find toolchain file:
  ../cmake/gcc-arm-none-eabi-8-2019-q3-update.cmake
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

marckleinebudde, 2022-06-12:
Which CAN USB adatper do you have?

Hemi940, 2022-06-12:
BIGTREETECH U2C 1.1
And BIGTREETECH EBB36 CAN V1.1

marckleinebudde, 2022-06-12:
BIGTREETECH don't have an official firmware in this repo. However their [datasheet](https://github.com/bigtreetech/U2C/blob/master/BIGTREETECH%20U2C%20V1.0%26V1.1%20User%20Manual.pdf) says they are using a STM32F072C8.

**Note: the following may brick your device!**

You can try flashing the candleLight firmware, which uses a STM32F072, too. Why do you want to upgrade? Are there any issues with the original firmware?

Hemi940, 2022-06-12:
In their instructions, it said use Candlight firmware.
Can i just build it with Make config and Make then use the bin file?
Also I'm would like to run The Octopus board trough the can as well.
Not to sure i understand the difference or the use between USB and Can for this adapter.
Thank You for your help.

marckleinebudde, 2022-06-12:
Why do you want to flash the device? I assume it comes with firmware.

Hemi940, 2022-06-12:
I thought it was required.

marckleinebudde, 2022-06-12:
What does `lsusb` show, if you attach the CAN-USB adapter to a Linux machine?

Hemi940, 2022-06-12:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapte                                                                                        r
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

marckleinebudde, 2022-06-12:
```
Bus 001 Device 003: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapte r
```

That's the device running with firmware. so it comes pre-flashed.

Hemi940, 2022-06-13:
when connect EBB36 V1.1 This is what i get.
Made config with Klipper and flashed device with klipper.bin
Hooked 24v to U2C board and attached 4 buss wires.
Any Thoughts? 

pi@raspberrypi:~ $ ~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0
Traceback (most recent call last):
  File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interfaces/socketcan/socketcan.py", line 679, in _send_once
    sent = self.socket.send(data)
OSError: [Errno 100] Network is down

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/klipper/scripts/canbus_query.py", line 64, in <module>
    main()
  File "/home/pi/klipper/scripts/canbus_query.py", line 61, in main
    query_unassigned(canbus_iface)
  File "/home/pi/klipper/scripts/canbus_query.py", line 25, in query_unassigned
    bus.send(msg)
  File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interfaces/socketcan/socketcan.py", line 658, in send
    sent = self._send_once(data, msg.channel)
  File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interfaces/socketcan/socketcan.py", line 681, in _send_once
    raise can.CanError("Failed to transmit: %s" % exc)
can.CanError: Failed to transmit: [Errno 100] Network is down

marckleinebudde, 2022-06-13:
Have you read the error message?
```
OSError: [Errno 100] Network is down
```

The `can0` network interface is down. You have to configure the bitrate (here 500 kbit/s) and bring the interface up:
```
ip link set can0 up type can bitrate 500000
```

fenugrec, 2022-08-27:
Closing for inactivity

closed 2022-08-27 by fenugrec