candle-usb/candleLight_fw #106: recent gcc complains about "LOAD segment with RWX permissions"

fenugrec, 2022-08-28:
`/usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: warning: cantact_fw has a LOAD segment with RWX permissions`

Trying the naive fix of editing the .ldscripts RAM section as

`RAM   (rw) : ORIGIN = 0x20000000, LENGTH = 6K`

has no effect. 

A search quickly indicates we're not the only ones seeing this :

https://stackoverflow.com/questions/73429929/gnu-linker-elf-has-a-load-segment-with-rwx-permissions-embedded-arm-project

I haven't had time to dig any further.
I'm also fairly certain this can be disregarded on baremetal, non-MMU firmware like this.

marckleinebudde, 2022-11-23:
FYI: https://lore.kernel.org/all/20220810222442.2296651-1-ndesaulniers@google.com

fenugrec, 2022-11-23:
can we add `--no-warn-rwx-segments`systematically or would that break on older binutils versions ?

fenugrec, 2022-11-23:
Maybe cmake has a mechanism to check for ld-options ? 
Projects using plain makefiles have to resort to this kind of script

https://github.com/OP-TEE/optee_os/blob/203ee23d005b2cec2f21b5de334c5a246be32599/mk/cc-option.mk

marckleinebudde, 2022-11-23:
https://gist.github.com/multiplemonomials/9cc2c4343a4bdd93689520ce64098b84