candle-usb/candleLight_fw #116: Handling GS_USB_BREQ_SET_TERMINATION and GS_USB_BREQ_GET_TERMINATION requests

fenugrec, 2022-10-12:
Continuing discussion from #113.

There is a difference in behaviour between old firmware, and new firmware compiled without TERM_PIN , when receiving those new requests.

Daniel-Trevitz, 2022-10-12:
I suspect that USBD_GS_CAN_EP0_RxReady will never be called as the USBD_GS_CAN_Config_Request drops it out first; I.E. the check there is uneeded.

unless you replace the param_u32 with a constant, the if statement will not be optimized out.
The memcpy can not be optimized out.

This is optimized out:
		case GS_USB_BREQ_SET_TERMINATION:
			set_term(req->wValue, *(uint32_t*)hcan->ep0_buf);
			break;

This is insufficient for setting an invalid terminating resistor when it's enabled.

found it. PR coming in hot

fenugrec, 2022-10-15:
Closed via pr #117, thanks

closed 2022-10-15 by fenugrec