candle-usb/candleLight_fw #140: startup.c is broken - hardfault with large BSS

fenugrec, 2022-11-17:


fenugrec, 2022-11-17:
incidentally, looking at disasm for good build, there's already a `_start` function that initializes BSS (__bss_start__ to __bss_end__) with `memset`. Still looking into what part of newlib-nano would handle initialized data - I think we have redundancy in startup.c that we could get rid of.

marckleinebudde, 2022-11-17:
I think the problem is the too large `data` section. See https://github.com/candle-usb/candleLight_fw/pull/141 for a fix.

fenugrec, 2022-11-17:
Fixed in 59896b65fdd588fed40cc4d9efec00d21f04ad31 via #141 . We still need to initialized .rodata because newlib will not do that for us (it does clear BSS though)

closed 2022-11-17 by fenugrec

marckleinebudde, 2022-11-17:
`.rodata` is read only data, right? Does it live in the ROM?

fenugrec, 2022-11-17:
> `.rodata` is read only data, right? Does it live in the ROM?

my bad, I meant the initialized RAM stuff, `.data`

marckleinebudde, 2022-11-17:
My patch replaced the 32-bit manual copy loop by a `memcpy()`, which takes length in bytes.

fenugrec, 2022-11-17:
>So everything should be fine?

Yes. We can't get rid of startup.c , but it should work now.

marckleinebudde, 2022-11-17:
FYI: https://github.com/ARM-software/CMSIS_5/issues/632