candle-usb/candleLight_fw #154: Feature: add support for GS_CAN_FEATURE_GET_STATE and GS_CAN_FEATURE_BERR_REPORTING

marckleinebudde, 2023-01-09:


marckleinebudde, 2023-01-09:
As far as I can see in https://github.com/candle-usb/candleLight_fw/pull/107 and https://github.com/candle-usb/candleLight_fw/pull/119 the error frame generation was cleaned up. I want to make sure that this is merged and no code was forgotten.

marckleinebudde, 2023-01-09:
In https://github.com/candle-usb/candleLight_fw/pull/124#issuecomment-1303920718 @jhofstee says his implementation can be rebased to mainline.

fenugrec, 2023-02-24:
I wonder if this could be extended to count dropped frames (RX overflows) ...

marckleinebudde, 2023-02-24:
We already have the `GS_CAN_FLAG_OVERFLOW` to signal a RX overflow to Linux:
https://github.com/candle-usb/candleLight_fw/blob/4a1e646cdb0b5b45bbc6c5bf77013205386f2da6/include/gs_usb.h#L83