candle-usb/candleLight_fw #164: build warning: "_close is not implemented and will always fail"

fenugrec, 2023-04-28:
Nice, new warnings with arm-gcc 13.1.0 :

```
/usr/lib/gcc/arm-none-eabi/13.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.1.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(libc_a-closer.o): in function `_close_r':
/build/arm-none-eabi-newlib/src/build-nano/arm-none-eabi/thumb/v7e-m/nofp/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/closer.c:47: warning: _close is not implemented and will always fail
/usr/lib/gcc/arm-none-eabi/13.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.1.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(libc_a-lseekr.o): in function `_lseek_r':
/build/arm-none-eabi-newlib/src/build-nano/arm-none-eabi/thumb/v7e-m/nofp/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/lseekr.c:49: warning: _lseek is not implemented and will always fail
/usr/lib/gcc/arm-none-eabi/13.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.1.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(libc_a-readr.o): in function `_read_r':
/build/arm-none-eabi-newlib/src/build-nano/arm-none-eabi/thumb/v7e-m/nofp/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/readr.c:49: warning: _read is not implemented and will always fail
/usr/lib/gcc/arm-none-eabi/13.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.1.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(libc_a-writer.o): in function `_write_r':
/build/arm-none-eabi-newlib/src/build-nano/arm-none-eabi/thumb/v7e-m/nofp/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/writer.c:49: warning: _write is not implemented and will always fail
```

those are neither implemented nor used. I'm not sure which step is pulling in those symbols; none of them end up in the compiled .elf.
Is it my distro's build that got something wrong all of a sudden or is anyone else also seeing this ?

fenugrec, 2023-04-28:
Some possible leads
https://stackoverflow.com/questions/73742774/gcc-arm-none-eabi-11-3-is-not-implemented-and-will-always-fail
https://stackoverflow.com/questions/75626690/arm-none-eabi-gcc-is-not-implemented-and-will-always-fail-messages

ryedwards, 2023-09-11:
The STMCUBEMX HAL added a "syscalls.c" to stub out these functions to prevent compiler errors:
https://github.com/ryedwards/budgetcan_fw/blob/main/Core/Src/syscalls.c

closed 2024-06-05 by fenugrec