candle-usb/candleLight_fw #169: Openmoko firmware upgrade , no jumper to enter DFU mode or button

pergolafabio, 2023-06-06:
Hi,

hope someone can help me, i have purchased an canable, seems its an Openmoko, i want to change firmware on it, was testing it like below, but it gives errors?
As you can see in screenshot, i dont have a jumper to enter DFU mode? nor do i have a button on it? Anyone seen this before?

[image]

[image]

marckleinebudde, 2023-06-06:
Use `make flash-candleLight_fw` to flash the firmware.

pergolafabio, 2023-06-06:
hi, thnx for answering, here is the output : 

```
debian@debian:~$ lsusb|grep -i open
Bus 002 Device 004: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
```

marckleinebudde, 2023-06-06:
> hi, thnx for answering, here is the output :
> 
> ```
> debian@debian:~$ lsusb|grep -i open
> Bus 002 Device 004: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
> ```

looks good!

marckleinebudde, 2023-06-06:
If it's a canable device better flash the canable firmware: `make flash-canable_fw`.

pergolafabio, 2023-06-06:
sorry, indeed new to this, but according to canable site, i also need to set the jumper or press the button, but i dont see any of those?

marckleinebudde, 2023-06-06:
It should work without an extra jumper.

pergolafabio, 2023-06-06:
ah ok, just unclear how to flash? i have downloaded the canable_fw file, as you can see? but the command gives an error?

[image]

codebuk, 2023-06-06:
I have a sample here of that board. It has a pair of BOOT0 pins labelled on
the rear. I have not reflashed this unit.

[image: image.png]

pergolafabio, 2023-06-06:
hmm, dont see your image

marckleinebudde, 2023-06-06:
try `dfu-util -a 0 -R -s 0x08000000 -D canable_fw.dfu`

codebuk, 2023-06-06:
[image]

pergolafabio, 2023-06-06:
yeah, mine looks the same indeed, it just named 303canable3 instead

i need to shortwire those pins then ? buts its a plastic box :-)

marckleinebudde, 2023-06-06:
open the plastic box, or use a solder iron to make some holes :)

pergolafabio, 2023-06-06:
omg, why did they  make it so difficult then

marckleinebudde, 2023-06-06:
Use some metal tweezers or unfold a paper clip.

pergolafabio, 2023-06-06:
just last question, if i use a tweezer or clip, do i need to connect it the while time?
Or is is enough to use the tweezer, then plugin usb , at that point , can i detach the tweezer?

closed 2023-06-06 by pergolafabio