candle-usb/candleLight_fw #171: version for stm32F042C4xx?

AlekseyMamontov, 2023-10-17:
Good afternoon,
Tell me, is there a firmware version for stm32F042C4xx?
Has anyone managed to fit a project into 16 kbytes? ))

marckleinebudde, 2023-10-17:
You can try v2.1 release https://github.com/candle-usb/candleLight_fw/releases/tag/v2.1, before we switched from `-Os` to `-O2`. It still fits into 16k:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       14720 B        32 KB     44.92%
             RAM:        3744 B         6 KB     60.94%
create and sign dfu bin file: cannette_fw
   text    data     bss     dec     hex filename
  14708     184    3572   18464    4820 cannette_fw
[100%] Built target cannette_fw
```

Use the `cannette` board a template.

fenugrec, 2023-10-17:
Unless you have already purchased a large number of these, I *strongly* recommend going with the next larger flash size (32k). Otherwise it may or may not fit, depending on compiler version, optimization, etc.

closed 2023-10-17 by fenugrec

AlekseyMamontov, 2023-10-18:
> You can try v2.1 release https://github.com/candle-usb/candleLight_fw/releases/tag/v2.1, before we switched from `-Os` to `-O2`. It still fits into 16k:
>
> ```
> Memory region         Used Size  Region [...]

Thank you, it worked with the firmware for cantcat

Linux
cannette_fw.bin
It is detected as a USB device and configured as CAN - but does not work.

CandleLight_fw.bin
Can't be seen as a USB device at all

Average size 14.5 kb

AlekseyMamontov, 2023-10-18:
the current version takes up even less than 13.8kb
with option
-Os

marckleinebudde, 2023-10-19:
You probably have to adopt the firmware for your board.

marckleinebudde, 2023-10-19:
> the current version takes up even less than 13.8kb with option `-Os`

We switched to `-O2`, as it's a means a considerable performance improvement. See commit message of 0612b5017f322d8675deb8a999312179ee437e48

AlekseyMamontov, 2023-10-19:
I understand, I’ll buy an STM32F042C6 a little later.
I just assembled a test version of the adapter board, and STM32F042C4 chips are in stock, it’s good that the crisis for chips is gradually going away).
The CAN network in my equipment is not heavily loaded.
I think there will be no missing packets.
As I understand it, -O3 will also add performance.
but not significantly.