candle-usb/candleLight_fw #173: CandleLight FD at 8 Mbps

tomasjakubik, 2023-12-22:
This issue is specifically with candleLight FD and [fork topic/candleLightFD](https://github.com/linux-automation/candleLight_fw/tree/topic/candleLightFD), but that doesn't allow issues.

We are starting a CAN FD project. We bought some candleLight FDs to debug and perhaps to manage the network from Linux. The issue is that this firmware doesn't compensate for transceiver delay and so it cannot transmit over about 2 Mbps. Receive works fine with 1 & 8 Mbps, but when you try transmitting, the FDCAN peripheral cannot verify the bits it sent, fails, repeats and fails again.
Without Tx delay compensation, my code was behaving in a similar way. Here is some code that works for me on G4 nucleo clocked at 160 MHz:

```c++
        hfdcan2.Instance = FDCAN2;
        hfdcan2.Init.ClockDivider = UINT32_MAX; // Divider is not available for FDCAN2
        hfdcan2.Init.FrameFormat = FDCAN_FRAME_FD_BRS;
        hfdcan2.Init.Mode = FDCAN_MODE_NORMAL;
        hfdcan2.Init.AutoRetransmission = ENABLE;
        hfdcan2.Init.TransmitPause = ENABLE;
        hfdcan2.Init.ProtocolException = ENABLE;
        hfdcan2.Init.NominalPrescaler = 1;
        hfdcan2.Init.NominalSyncJumpWidth = 32;
        hfdcan2.Init.NominalTimeSeg1 = 127;
        hfdcan2.Init.NominalTimeSeg2 = 32;
        hfdcan2.Init.DataPrescaler = 1;
        hfdcan2.Init.DataSyncJumpWidth = 4;
        hfdcan2.Init.DataTimeSeg1 = 15;
        hfdcan2.Init.DataTimeSeg2 = 4;
        hfdcan2.Init.StdFiltersNbr = 0;
        hfdcan2.Init.ExtFiltersNbr = 0;
        hfdcan2.Init.TxFifoQueueMode = FDCAN_TX_QUEUE_OPERATION;
        if (HAL_FDCAN_Init(&hfdcan2) != HAL_OK) {
            Error_Handler();
        }

        // Set TIM3 as source of the timestamp
        if (HAL_FDCAN_EnableTimestampCounter(&hfdcan2, FDCAN_TIMESTAMP_EXTERNAL) != HAL_OK) {
            Error_Handler();
        }

        /// @note Configure Tx delay compensation of the used transceiver for high speed data phase.
        ///   FDCAN automatically measures delay between falling edge on FDCAN_TX and FDCAN_RX.
        ///
        /// TdcOffset:
        ///   This is a value to be added to the measured delay. FDCAN measures delay between falling edges.
        ///   TdcOffset is added to the measured delay to get in the middle of the bit.
        ///   Documentation sucks. Nobody knows what minimal time quanta is. More info could be probably in CiA 601-2 which is paywalled.
        ///   Seems like minimal time quanta is the master CAN clock. Note that CAN2 doesn't have divider.
        ///
        ///   The value used was found experimentally for 1 & 8 Mbps and TJA1051/T3 transceiver.
        ///   Delay between FDCAN_TX and FDCAN_RX measured by logic analyzer is approximately 100 ns.
        ///   Values from 5 to 20 worked. At 160 MHz the length 20 * 6.25 ns = 125 ns or one bit.
        ///   Perhaps the offset needs to be between 25% and 100% of the bit length.
        ///   Of the working interval, 75% was used (80% of the bit); TdcOffset = 16.
        ///
        /// TdcFilter:
        ///   This is a minimal value for the measured delay. It prevents glitch on Rx to trigger too early.
        ///
        ///   The value used was found experimentally for 1 & 8 Mbps and TJA1051/T3 transceiver.
        ///   When the transceiver is working at correct speed, the value from PSR/TDCV register was read; TDCV = 30.
        ///   Of that value, 75% was used; TdcFilter = 22.
        if (HAL_FDCAN_ConfigTxDelayCompensation(&hfdcan2, 16, 22) != HAL_OK) {
            Error_Handler();
        }
        if (HAL_FDCAN_EnableTxDelayCompensation(&hfdcan2) != HAL_OK) {
            Error_Handler();
        }
```

This is not ready to be a pull request. It will need different values at different speeds. Unfortunately, the documentation I have available is not good enough to have a proper solution.
When I have Tag-Connect I may have a more deeper look into the firmware, but I cannot promise anything.

I guess that you are in process of merging the fork here, but any ideas on how or even if it should be incorporated into the firmware?

marckleinebudde, 2024-01-09:
Hey @tomasjakubik,

I've cherry-picked and squashed @pgreenland's commit https://github.com/pgreenland/candleLight_fw/commit/4051608a60009072e7d2644a57f0802db70a135b into my https://github.com/marckleinebudde/candleLight_fw/tree/multichannel branch. Can you test, if this fixes your problem?

Cc: @lichtfeind

glinders, 2025-10-06:
Yes, you are right! I missed that TDCO is expressed in **mtq** units and not **tq** units. On the hardware I tested, the data BRP was always 1, so it didn't make a difference.

So the correct computation is:   `TDCO = ( 1 + PROP_SEG + PHASE_SEG1 ) × DATA_PRESCALER `

For STM32, this translates to:   
```c
uint32_t tdco = (1 + channel->channel.Init.DataTimeSeg1) * channel->channel.Init.DataPrescaler;

HAL_FDCAN_ConfigTxDelayCompensation(&channel->channel, tdco, 0U);
```

See `2.3.15 Transmitter Delay Compensation Register (TDCR)` on page 19 of:   https://www.bosch-semiconductors.com/media/ip_modules/pdf_2/m_can/mcan_users_manual_v331.pdf

(The STM32 MCUs with FDCAN peripherals use the Bosch M-CAN IP-module. Same for most CAN FD capable MCUs of NXP, TI and many other manufacturers.)

Zephyr OS also uses the same implementation:
```c
/**
 * @brief Calculate Transmitter Delay Compensation Offset from data phase timing parameters.
 *
 * Calculates the TDC Offset in minimum time quanta (mtq) using the sample point and CAN core clock
 * prescaler specified by a set of data phase timing parameters.
 *
 * The result is clamped to the minimum/maximum supported TDC Offset values provided.
 *
 * @param _timing_data Pointer to data phase timing parameters.
 * @param _tdco_min    Minimum supported TDC Offset value in mtq.
 * @param _tdco_max    Maximum supported TDC Offset value in mtq.
 * @return             Calculated TDC Offset value in mtq.
 */
#define CAN_CALC_TDCO(_timing_data, _tdco_min, _tdco_max)                                          \
	CLAMP((1U + _timing_data->prop_seg + _timing_data->phase_seg1) * _timing_data->prescaler,  \
	      _tdco_min, _tdco_max)
```

Elmue, 2025-10-15:
@pgreenland

You are perfectly correct with all what you write.

>  I stopped at that point trying to understand why it was adjustable in that case.

That is a very good question which I was also asking myself.
If the processor already measures the delay alone, why do we have to configure this SSP at all ?

I asked in the forum of ST but there is nobody who has a clue about their own products that they sell.
But at least they recommended me the PDF documents, that I uploaded, which are very informative.

Finally I still had remaining questions and they told me to write an email to the ST support - what I did.
The answer from the support was a huge disillusion.
The guy told me to read the reference manual.
That is the so called "support" that we get from ST.
They have employees who have no idea about these details.
So we will never get an answer to this question from ST.

> that the two edges might not be symmetric in nature for a particular transceiver.

This sounds like a reasonable explanation.

No matter at the end, I tested my code with all possible data baudrates (1, 2, 2.5, 4, 5, 8 and 10 Mbit)  and it works perfectly.