candle-usb/candleLight_fw #174: CPU_FAMILY STM32G0B1XK vs. CPU_FAMILY STM32G0B1XE (possible bug)

hcet14, 2023-12-23:
I'm new with candleLight_fw and try to understand parts of the code. Thank you all, great work!

With the recently introduced support of STM32G0B1 you named the CPU family "STM32G0B1XK".

Looking at STM32F042X6, STM32F072XB, and STM32F407XE and trying to understand why they have those cryptic names, I stumbled over the "Ordering information" in the datasheets. Now it made sense to me.
[image]
Shouldn't the family be named "STM32G0B1XE"?

marckleinebudde, 2023-12-23:
> Shouldn't the family be named "STM32G0B1XE"?

The trailing "E", i.e. the flash memory size isn't important for us, so "STM32G0B1X" should be sufficient. Feel free to create a pull request to rename the CPU family in the project.

fenugrec, 2023-12-23:
hm, I hadn't look closely at that, but at first glance indeed I'm not sure why there is a 'K' there. 
Our other targets have an 'x' there because the pincount may change, but the flash size must be at least 32k for the fw to fit. We also generate an appropriate ldscript based on that flash size.

But now you have me wondering if we need to know the exact flash size at all, as long as it's above a certain minimum. Pincount never matters anyway since the board def takes care of assigning pins.

hcet14, 2023-12-24:
Thanks for your replies.

I'm no programmer at all. I try to build "https://github.com/marckleinebudde/candleLight_fw/tree/multichannel" branch. After some tries it built, but I'm just a beginner and never built a github repository :wink: 
The final goal is, getting Klipper running using CAN-FD with a toolhead board. Home-made 3D printers tend to produce a lot of noise, so CAN-FD might be a good way to follow.

Trying to understand parts of the code as a complete rookie makes it harder if "things" aren't easy comprehensible. 
If you follow a strict concept, for naming variables, names, etc, to make it easier to comprehend for a newbie like me is also helping you.

> The trailing "E", i.e. the flash memory size isn't important for us, so "STM32G0B1X" should be sufficient.

Of course. I won't open a pull request. I renamed this issue.

> But now you have me wondering if we need to know the exact flash size at all, as long as it's above a certain minimum.

Just the 042 sub-family might run into a problem. The minimum user code memory size is 16 Kbyte, all other supported sub-families have at least 64 Kbyte. Your list of supported boards with 042 have all 32 Kbyte MCUs.

[image]

Why not naming the CPU families just with "Device family-Product type-Sub-family" like "STM32G0B1" or "STM32F042"? That might be worth opening a pull request.

closed 2024-04-02 by hcet14