candle-usb/candleLight_fw #175: Potential firmware / driver issue

luckylinux, 2024-01-09:
I'm using some CanAble Pro adapters I bought off AliExpress and I could communicate with the devices connected to them initially.

Since then, not sure if related to upgrading Debian Bullseye to Debian Bookworm, Firmware BUG / incompatibility, or what else, my program using python-can says it sent the CAN message correctly, yet nothing appears in `ifconfig`and the device does not react at all. `candump` doesn't log anything either.

Is there a way to check the current version of the firmware ? I'm not even sure these are STM32 microcontroller based ...

dfu-util
```
dfu-util -l -v
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

libusb version 1.0.26 (11724)
Found Runtime: [1d50:606f] ver=0000, devnum=6, cfg=1, intf=1, path="1-1.2.5", alt=0, name="candleLight firmware upgrade interface", serial="003D00474E56510220373334"
Found Runtime: [1d50:606f] ver=0000, devnum=7, cfg=1, intf=1, path="1-1.2.7", alt=0, name="candleLight firmware upgrade interface", serial="002A002D4730511420303650"
```

dmesg can
```
dmesg | grep -i can
[    6.060750] usb 1-1.2.5: Product: candleLight USB to CAN adapter
[    6.350431] usb 1-1.2.7: Product: candleLight USB to CAN adapter
[   13.570422] CAN device driver interface
[   14.755824] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[   14.758447] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[   17.209193] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   18.232968] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[   80.618222] can: controller area network core
[   80.618297] NET: Registered PF_CAN protocol family
[   80.633199] can: raw protocol
```

dmesg usb
```
dmesg ¬ grep -i usb
Try 'dmesg --help' for more information.
root@SolarMonitor:/opt/solar-controller/charger# dmesg | grep -i usb
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  usbcore.autosuspend=-1 console=ttyAMA0,115200 console=tty1 root=PARTUUID=24485673-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DK
[    0.130337] usbcore: registered new interface driver usbfs
[    0.130410] usbcore: registered new interface driver hub
[    0.130479] usbcore: registered new device driver usb
[    0.130709] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    2.186821] usbcore: registered new interface driver lan78xx
[    2.189407] usbcore: registered new interface driver smsc95xx
[    3.172220] dwc_otg 3f980000.usb: DWC OTG Controller
[    3.174397] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
[    3.176626] dwc_otg 3f980000.usb: irq 89, io mem 0x00000000
[    3.183203] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    3.187477] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    3.189742] usb usb1: Product: DWC OTG Controller
[    3.191973] usb usb1: Manufacturer: Linux 6.1.0-rpi7-rpi-v7 dwc_otg_hcd
[    3.194281] usb usb1: SerialNumber: 3f980000.usb
[    3.197171] hub 1-0:1.0: USB hub found
[    3.202491] usbcore: registered new interface driver usb-storage
[    3.223804] usbcore: registered new interface driver usbhid
[    3.226123] usbhid: USB HID core driver
[    3.533042] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    3.783286] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
[    3.793823] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.824944] hub 1-1:1.0: USB hub found
[    4.223063] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    4.943509] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
[    4.957971] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    5.104863] SMSC LAN8700 usb-001:003:01: attached PHY driver (mii_bus:phy_addr=usb-001:003:01, irq=199)
[    5.127546] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:ad:01:4e
[    5.243134] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[... 185 lines cut ...]
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      1 Device Firmware Update
      bInterfaceProtocol      1
      iInterface            224 candleLight firmware upgrade interface
      Device Firmware Upgrade Interface Descriptor:
        bLength                             9
        bDescriptorType                    33
        bmAttributes                       11
          Will Detach
          Manifestation Intolerant
          Upload Supported
          Download Supported
        wDetachTimeout                    255 milliseconds
        wTransferSize                    2048 bytes
        bcdDFUVersion                   1.1a
Device Status:     0x0000
  (Bus Powered)
```

It's like the CAN device is properly recognized but it's doing absolutely nothing ...

marckleinebudde, 2024-01-09:
What's the output of `ip --details -s -s -s link show can0`, after booting and configuring you CAN interface.

luckylinux, 2024-01-09:
Not sure what you mean by "configuring your CAN" interface.

1. I reboot the Raspberry Pi.
2. Without even configuring the interface I get
```
3: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0  allmulti 0 minmtu 0 maxmtu 0 
    can state STOPPED restart-ms 0 
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp_inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 parentbus usb parentdev 1-1.2.5:1.0 
    RX:   bytes packets errors dropped  missed   mcast           
              0       0      0       0       0       0 
    RX errors:   length    crc   frame    fifo overrun
                      0      0       0       0       0 
    TX:   bytes packets errors dropped carrier collsns           
              0       0      0       0       0       0 
    TX errors:  aborted   fifo  window heartbt transns
                      0      0       0       0       0 
```
3. After `ip link set can0 type can bitrate 125000 restart-ms 1500` I get
```
3: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0  allmulti 0 minmtu 0 maxmtu 0 
    can state STOPPED restart-ms 1500 
	  bitrate 125000 sample-point 0.875
	  tq 500 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1 brp 24
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp_inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 parentbus usb parentdev 1-1.2.5:1.0 
    RX:   bytes packets errors dropped  missed   mcast           
              0       0      0       0       0       0 
    RX errors:   length    crc   frame    fifo overrun
                      0      0       0       0       0 
    TX:   bytes packets errors dropped carrier collsns           
              0       0      0       0       0       0 
    TX errors:  aborted   fifo  window heartbt transns
                      0      0       0       0       0 

```
4. After `ip link set can0 type can bitrate 125000 restart-ms 1500` and `ip link set up can0` I get
```
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0  allmulti 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 1500 
	  bitrate 125000 sample-point 0.875
	  tq 500 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1 brp 24
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp_inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 parentbus usb parentdev 1-1.2.5:1.0 
    RX:   bytes packets errors dropped  missed   mcast           
              0       0      0       0       0       0 
    RX errors:   length    crc   frame    fifo overrun
                      0      0       0       0       0 
    TX:   bytes packets errors dropped carrier collsns           
              0       0      0       0       0       0 
    TX errors:  aborted   fifo  window heartbt transns
                      0      0       0       0       1 
```

Duh ... There is a TX error with 1 for transmitted (bottom right in the output) ?

luckylinux, 2024-01-09:
> > Duh ... There is a TX error with 1 for transmitted (bottom right in the "table") ?
> 
> I see the same error value of "1" for my candlelight adapter, as well as for another USB CAN adapter. Don't know where this comes from. However the rest looks fine.
> 
> Now configure `can0` and `can1` and connect both. Add a termination on each end of the bus. Run:
> 
> ```
> candump any,0:0,#FFFFFFFF -cexdtA &
> cansend can0 002#8877665544332211
> cansend can1 002#1122334455667788
> ```

What do you mean exactly ? Right now each of them are (in the garage) connected to one separate Emerson R48-3000e Rectifier.

Do you want me to connect the can0 and can1 adapter between themselves in a "loop" (can0 CAN_H goes to can1 CAN_H, can0 CAN_L goes to can1 CAN_L) ? I don't think I'll have time to go into the garage and reassembly everything before the weekend :(.

marckleinebudde, 2024-01-09:
> What do you mean exactly ? Right now each of them are (in the garage) connected to one separate Emerson R48-3000e Rectifier.

OK

> Do you want me to connect the can0 and can1 adapter between themselves in a "loop" (can0 CAN_H goes to can1 CAN_H, can0 CAN_L goes to can1 CAN_L) ? I don't think I'll have time to go into the garage and reassembly everything before the weekend :(.

Exactly make a small CAN bus with `can0` and `can1`. Yes, all CAN_L are connected to each other and all CAN_H are connected. You need a 120 ohm termination resistor at both ends of the CAN bus, see https://www.ni.com/en/support/documentation/supplemental/09/can-physical-layer-and-termination-guide.html

fenugrec, 2024-01-09:
Who knows what version of candlelight those are running - we added the git commit hash to the iConfiguration string in april 2021. I don't know if the other forks followed suit...
You could try making a backup first with dfu-util then trying a more recent build.

luckylinux, 2024-01-09:
Not sure if that helps, but I got three status LED indicators and they all seem to be constantly ON.

[image]

But yes, I could try with a more recent version. It's just very weird that it worked in the beginning (possibly with an older version of Debian / Linux Kernel).

marckleinebudde, 2024-01-09:
Looks like [these](https://ae01.alicdn.com/kf/Sb4c7f3325a734b5fb1620d7ba401651ai.jpg). Can you see which STM32 these are using?

luckylinux, 2024-01-09:
This was the listing: https://www.aliexpress.com/item/1005004972158302.html?spm=a2g0o.order_list.order_list_main.5.38c71802xenhek

I tried asking the seller if they are using STM32 and which one if so ...

marckleinebudde, 2024-01-09:
Have a look at the µC, it's the chip with the legs on all 4 sides.

luckylinux, 2024-01-09:
Yeah I thought so. The square one LFQP or something like that ...
I'll check this weekend I guess, but usually they scrape the part numbers out or paint it so you cannot reverse engineer (not that it's needed in this case since the project is open source, but you never know ...)

fenugrec, 2024-01-09:
Well the dfu-util output at least seems to match what an stm32 would show. just try `dfu-util -U temp.bin`,  will probably fail but place the device in DFU mode, then `dfu-util -l`, then -U again

luckylinux, 2024-01-09:
Not sure, the second -U fails, and now there are 2 devices (well total is 3, but I only tried to update 1 - I have 2 CanAble Pro plugged in):

```
root@SolarMonitor:/usr/src/candlelight_fw/20240109_Test_Dump_FW# dfu-util -U temp.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device

root@SolarMonitor:/usr/src/candlelight_fw/20240109_Test_Dump_FW# dfu-util -S "003D00474E56510220373334" -U temp.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 1d50:606f
Run-Time device DFU version 011a
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
dfu-util: error detaching
Device will detach and reattach...
dfu-util: Lost device after RESET?
root@SolarMonitor:/usr/src/candlelight_fw/20240109_Test_Dump_FW# dfu-util -l
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=8, cfg=1, intf=0, path="1-1.2.5", alt=1, name="@Option Bytes  /0x1FFFF800/01*016 e", serial="FFFFFFFEFFFF"
Found DFU: [0483:df11] ver=2200, devnum=8, cfg=1, intf=0, path="1-1.2.5", alt=0, name="@Internal Flash  /0x08000000/064*0002Kg", serial="FFFFFFFEFFFF"
Found Runtime: [1d50:606f] ver=0000, devnum=7, cfg=1, intf=1, path="1-1.2.7", alt=0, name="candleLight firmware upgrade interface", serial="002A002D4730511420303650"
root@SolarMonitor:/usr/src/candlelight_fw/20240109_Test_Dump_FW# dfu-util -S "003D00474E56510220373334" -U temp.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: No DFU capable USB device available
root@SolarMonitor:/usr/src/candlelight_fw/20240109_Test_Dump_FW# dfu-util -U temp.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device

root@SolarMonitor:/usr/src/candlelight_fw/20240109_Test_Dump_FW# dfu-util -S "FFFFFFFEFFFF" -U temp.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device

```

fenugrec, 2024-01-09:
Disconnect the other and try `dfu-util -a 0 -U ...`

luckylinux, 2024-01-09:
Well then I have to try during the weekend. It's in the garage so ...

luckylinux, 2024-01-09:
Should I just reboot now to "free it" for now ?

fenugrec, 2024-01-09:
> Well then I have to try during the weekend. It's in the garage so ...

ah, well you can play with dfu-util options to help it select the proper interface by device path (not sure why your attempt with `-S` didn't work). And yes reboot will exit DFU mode

luckylinux, 2024-01-09:
The attempt with -S worked the first time. But then 2 devices got created with the same FFFFFFFEFFFF serial number:
```
Found DFU: [0483:df11] ver=2200, devnum=8, cfg=1, intf=0, path="1-1.2.5", alt=1, name="@Option Bytes  /0x1FFFF800/01*016 e", serial="FFFFFFFEFFFF"
Found DFU: [0483:df11] ver=2200, devnum=8, cfg=1, intf=0, path="1-1.2.5", alt=0, name="@Internal Flash  /0x08000000/064*0002Kg", serial="FFFFFFFEFFFF"
```

Whereas the untouched CanAble Pro adapter still has its previous serial number:
```
Found Runtime: [1d50:606f] ver=0000, devnum=7, cfg=1, intf=1, path="1-1.2.7", alt=0, name="candleLight firmware upgrade interface", serial="002A002D4730511420303650"
```

fenugrec, 2024-01-09:
yes, the first time causes the stm32 to enter native DFU mode which exposes two interfaces, alt=0 and alt=1, and I see now they have the same FFFEFFF serial which doesn't help to disambiguate which one to use. You want the "internal flash" to grab the firmware.

luckylinux, 2024-01-12:
OK quick try in the garage today.

Removed power supply (AC adapter) from USB Hub. Now it's passive.

Unplugged & replugged in the CanAble Pro Adapter. Launched my script.

Set the rectifier to 56V. Works like a charm.

So I don't know if the Powered USB hub is the root cause of the problem, but NO amount of rebooting the Raspberry Pi would fix the issue.

hcet14, 2024-04-02:
> Have a look at the µC, it's the chip with the legs on all 4 sides.

I'm quite sure the boards use STM32F072C8T6 µCs.