candle-usb/candleLight_fw #178: Problem building firmware for STM32G0B1

v0nHoFF, 2024-02-01:
Hello everyone! I'm struggling to compile the firmware for STM32G0B1 (CandleLight-FD) but when running "make help", there's no flash-STM32G0*_fw.

This is what make help returns:

hoff@hoff-VM:~/Downloads/a/candleLight_fw/build$ make help
The following are some of the valid targets for this Makefile:
... all (the default if no target is provided)
... clean
... depend
... edit_cache
... rebuild_cache
**... flash-CANable_MKS_fw
... flash-CONVERTDEVICE_xCAN_fw
... flash-DSD_TECH_SH_C30A_fw
... flash-FYSETC_UCAN_fw
... flash-STM32F4_DevBoard_fw
... flash-canable_fw
... flash-canalyze_fw
... flash-candleLight_fw
... flash-cannette_fw
... flash-cantact_fw
... flash-usb2can_fw**
... version_h
... CANable_MKS_fw
... CONVERTDEVICE_xCAN_fw
... DSD_TECH_SH_C30A_fw
... FYSETC_UCAN_fw
... STM32F4_DevBoard_fw
... STM32_HAL_STM32F042x6
... STM32_HAL_STM32F072xB
... STM32_HAL_STM32F407xE
... STM32_HAL_STM32G0B1xK
... STM32_USB_Device_Library_STM32F042x6
... STM32_USB_Device_Library_STM32F072xB
... STM32_USB_Device_Library_STM32F407xE
... STM32_USB_Device_Library_STM32G0B1xK
... canable_fw
... canalyze_fw
... candleLight_fw
... cannette_fw
... cantact_fw
... usb2can_fw
... src/can.obj
... src/can.i
... src/can.s
... src/device/device_f0.obj
... src/device/device_f0.i
... src/device/device_f0.s
... src/device/device_f4.obj
... src/device/device_f4.i
... src/device/device_f4.s
... src/dfu.obj
... src/dfu.i
... src/dfu.s
... src/gpio.obj
... src/gpio.i
... src/gpio.s
... src/interrupts.obj
... src/interrupts.i
... src/interrupts.s
... src/led.obj
... src/led.i
... src/led.s
... src/main.obj
... src/main.i
... src/main.s
... src/startup.obj
... src/startup.i
... src/startup.s
... src/timer.obj
... src/timer.i
... src/timer.s
... src/usbd_conf.obj
... src/usbd_conf.i
... src/usbd_conf.s
... src/usbd_desc.obj
... src/usbd_desc.i
... src/usbd_desc.s
... src/usbd_gs_can.obj
... src/usbd_gs_can.i
... src/usbd_gs_can.s
... src/util.obj
... src/util.i
... src/util.s

marckleinebudde, 2024-02-01:
STM32G0B1-based devices are not yet supported by the mainline firmware. Support for these devices is discussed in #139 and #176.

Which branch are you using?

karel1344, 2024-02-15:
> STM32G0B1-based devices are not yet supported by the mainline firmware. Support for these devices is discussed in #139 and #176.
> 
> Which branch are you using?

Hi,
I'm also trying to build firmware for candleLight FD at your branch (https://github.com/marckleinebudde/candleLight_fw/tree/multichannel)

However, when I try to build, I get an error as follows

`make candleLightFD_fw`

```
[  7%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal.c.obj
[  7%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_fdcan.c.obj
[  7%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_cortex.c.obj
[  7%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_flash_ex.c.obj
[ 15%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_flash.c.obj
[ 15%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_gpio.c.obj
[ 15%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_pcd_ex.c.obj
[ 23%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_pcd.c.obj
[ 23%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_rcc.c.obj
[ 23%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_rcc_ex.c.obj
[ 30%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_pwr.c.obj
[ 30%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_pwr_ex.c.obj
[ 30%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_tim_ex.c.obj
[ 38%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_hal_tim.c.obj
[ 38%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/stm32g0xx/stm32g0xx_ll_usb.c.obj
[ 38%] Building C object libs/STM32_HAL/CMakeFiles/STM32_HAL_STM32G0B1xK.dir/src/cmsis/system_stm32g0xx.c.obj
[ 46%] Linking C static library libSTM32_HAL_STM32G0B1xK.a
[ 46%] Built target STM32_HAL_STM32G0B1xK
[ 53%] Building C object libs/STM32_USB_Device_Library/CMakeFiles/STM32_USB_Device_Library_STM32G0B1xK.dir/Core/Src/usbd_ctlreq.c.obj
[ 53%] Building C object libs/STM32_USB_Device_Library/CMakeFiles/STM32_USB_Device_Library_STM32G0B1xK.dir/Core/Src/usbd_ioreq.c.obj
[ 53%] Building C object libs/STM32_USB_Device_Library/CMakeFiles/STM32_USB_Device_Library_STM32G0B1xK.dir/Core/Src/usbd_core.c.obj
[ 61%] Linking C static library libSTM32_USB_Device_Library_STM32G0B1xK.a
[ 61%] Built target STM32_USB_Device_Library_STM32G0B1xK
-- Found Git: /usr/bin/git (found version "2.17.1") 
[ 61%] Built target version_h
[ 61%] Building C object CMakeFiles/candleLightFD_fw.dir/src/usbd_desc.c.obj
[ 61%] Building C object CMakeFiles/candleLightFD_fw.dir/src/usbd_gs_can.c.obj
[ 61%] Building C object CMakeFiles/candleLightFD_fw.dir/src/usbd_conf.c.obj
[ 69%] Building C object CMakeFiles/candleLightFD_fw.dir/src/can_common.c.obj
[ 69%] Building C object CMakeFiles/candleLightFD_fw.dir/src/dfu.c.obj
[ 69%] Building C object CMakeFiles/candleLightFD_fw.dir/src/gpio.c.obj
[ 76%] Building C object CMakeFiles/candleLightFD_fw.dir/src/led.c.obj
[ 76%] Building C object CMakeFiles/candleLightFD_fw.dir/src/timer.c.obj
[ 76%] Building C object CMakeFiles/candleLightFD_fw.dir/src/util.c.obj
[ 84%] Building C object CMakeFiles/candleLightFD_fw.dir/src/startup.c.obj
[ 84%] Building C object CMakeFiles/candleLightFD_fw.dir/src/main.c.obj
[ 84%] Building C object CMakeFiles/candleLightFD_fw.dir/src/interrupts.c.obj
[ 92%] Building C object CMakeFiles/candleLightFD_fw.dir/src/boards/g0b1-candleLightFD.c.obj
[ 92%] Building C object CMakeFiles/candleLightFD_fw.dir/src/can/m_can.c.obj
/home/karel1344/candleLight_fw/src/can/m_can.c:59:21: error: initializer element is not constant
  .device_bt_const = CAN_btconst,
                     ^~~~~~~~~~~
/home/karel1344/candleLight_fw/src/can/m_can.c:59:21: note: (near initialization for 'CAN_btconst_ext.<anonymous>.device_bt_const')
CMakeFiles/candleLightFD_fw.dir/build.make:257: recipe for target 'CMakeFiles/candleLightFD_fw.dir/src/can/m_can.c.obj' failed
make[3]: *** [CMakeFiles/candleLightFD_fw.dir/src/can/m_can.c.obj] Error 1
CMakeFiles/Makefile2:957: recipe for target 'CMakeFiles/candleLightFD_fw.dir/all' failed
make[2]: *** [CMakeFiles/candleLightFD_fw.dir/all] Error 2
CMakeFiles/Makefile2:964: recipe for target 'CMakeFiles/candleLightFD_fw.dir/rule' failed
make[1]: *** [CMakeFiles/candleLightFD_fw.dir/rule] Error 2
Makefile:501: recipe for target 'candleLightFD_fw' failed
make: *** [candleLightFD_fw] Error 2
```

What did I miss?
I always appreciate your work.
Thank you.

marckleinebudde, 2024-02-15:
Which C compiler and which version are you using?

marckleinebudde, 2024-02-15:
Can you use a newer compiler?

karel1344, 2024-02-15:
> Can you use a newer compiler?

Can you recommend version? Even I changed gcc version to 11.4.0, still error occurred.

marckleinebudde, 2024-02-15:
I'm using `gcc version 13.2.1 20231009 (15:13.2.rel1-1)` without problems.

marckleinebudde, 2024-02-15:
I've worked around the issue, please try again: https://github.com/marckleinebudde/candleLight_fw/tree/multichannel

karel1344, 2024-02-19:
> I've worked around the issue, please try again: https://github.com/marckleinebudde/candleLight_fw/tree/multichannel

It works now!
Thank you.

v0nHoFF, 2024-03-28:
> STM32G0B1-based devices are not yet supported by the mainline firmware. Support for these devices is discussed in #139 and #176.
> 
> 
> 
> Which branch are you using?

I was using the main branch. I've switched to the FD one and it works. Thank you for the hint!

marckleinebudde, 2024-03-29:
Can you close the issue?

closed 2024-03-29 by v0nHoFF