candle-usb/candleLight_fw #186: Setting CAN auto retransmission

SergeTauger, 2024-10-04:
Hi all,
Could you please tell if auto retransmission in case no ACK received is enabled by default? If not, how can I enable it?
I searched the code but still have no idea which behavior listed in gs_usb.h is applied by default in can_enable.
Best, 
Serge

fenugrec, 2024-10-04:
that is a function of the hardware periphal ("NART" ) and "one shot mode" of the gs_usb driver. See https://github.com/candle-usb/candleLight_fw/blob/master/src/can/bxcan.c#L108

SergeTauger, 2024-10-04:
I've seen it, but can_enable is called once, in [usbd_gs_can.c/USBD_GS_CAN_EP0_RxReady
](https://github.com/candle-usb/candleLight_fw/blob/4ae1a7c2ad06c27d9d10a52a9e28614bab21e430/src/usbd_gs_can.c#L529) which is a callback. I cannot find where mode->flags is set and if is it set to |= one shot flag.

SergeTauger, 2024-10-04:
sorry, from the beginning of the thread I generated XY problem. Let me go into more details. Currently I'm working on CanOpen device (CanOpenNode port), my usb-can adapter is Fysetc UCan which runs candlelight_fw. CanOpen software on PC side is Lely CanOpen. Using candump with options "-d -D -e -x -t d" (capture all frames, show relative time) I get such a log
```
(000.946861)  can0  TX - -  605   [8]  2F 11 60 02 C0 00 00 00 
(000.000070)  can0  TX - -  605   [8]  23 12 60 02 88 13 00 00
 (000.000080)  can0  TX - -  605   [8]  2B 13 60 02 F4 01 00 00
 (000.000755)  can0  TX - -  605   [8]  2B 14 60 02 F4 01 00 00
 (000.000990)  can0  TX - -  605   [8]  2B 15 60 02 64 00 00 00
 (000.000925)  can0  TX - -  605   [8]  2F 10 60 02 21 00 00 00
 (000.001026)  can0  RX - -  585   [8]  60 11 60 02 00 00 00 00
 (000.001939)  can0  RX - -  585   [8]  60 13 60 02 00 00 00 00
 (000.001946)  can0  RX - -  585   [8]  60 14 60 02 00 00 00 00
 (000.001876)  can0  RX - -  585   [8]  60 15 60 02 00 00 00 00
```
As you can see, I write to device 6 times and get only 4 responces. As far as I can get it UCan should have resent a TX responce since it was not ACKed properly, but it didn't. Now I'm trying to get why.

fenugrec, 2024-10-04:
Are you sure you're not confusing canopen-level semantics with CAN ACK mechanism ? They are not the same, and with what you posted I see no indication that the raw CAN frames were not sent properly.

closed 2024-10-06 by fenugrec