candle-usb/candleLight_fw #188: sorry,iam a rookie，Who triggered this vendor packet?

tonghiden, 2025-02-19:
Who triggered this vendor packet?Why does this trigger as soon as I open the software?

[image]

[image]

marckleinebudde, 2025-02-19:
Which Software are you opening?

tonghiden, 2025-02-19:
Yes, I opened the Cangaroo software because I ported the descriptors to another chip platform (gd32), but my ported code did not receive the vendor package. Why

[image]

marckleinebudde, 2025-02-19:
What software did you actually port to the gd32?

marckleinebudde, 2025-02-19:
You mean https://github.com/candle-usb/candleLight_fw/tree/v1.0 ? This is 9 years... :/

You may want to diff the frames with the original firmware and your gd32 port and see if there's something different before this.

If not, then there's probably something wrong with your device setup.

tonghiden, 2025-02-20:
Hahaha, forgot, didn't you? Yes, it's been too long.But if you can remember, please help me see if there is any mark missing, because I didn't receive the vendor package when I opened the cangroo.exe on my own device, https://github.com/candle-usb/candleLight_fw/tree/v1.0 Yes, so, I wonder if some tags are missing so that the software can't recognise me, so they don't send me the vendor package, you know why?Also, I've got Windows to load up.，ths

[image]

tonghiden, 2025-02-20:
[image]

[image]
In addition, it is possible that a certain software has only been used in my country. I also tried it under Ubuntu and found that it couldn't form can0, and then I typed dmesg and found that it described [49401.290509] gs_usb 2-2.1:1.0: Couldn't send data format (err=-110)

fenugrec, 2025-02-20:
Do I understand correctly that you're using the 1.0 tree, which is ancient and very different from current codebase; ported it to a different platform, are not even showing your code, and expect us to provide support ?

tonghiden, 2025-02-20:
[image]

[image]
Sorry, maybe my description is wrong, I will re-describe. I bought a device with a chip STM32F07X in the shop, and then I compiled the source code of Candlelight1.0 and burned it to the device I purchased. I then used a Windows software called Cangroo; all the functions are normal, the message sending and receiving are perfect, and "can0" can also be loaded on Ubuntu. This proves that the code is compatible with the device (picture 1). Then I started to move Candlelight1.0 to another device (gd32/ch32), and when I thought the porting was complete, I found that Windows recognised the device and could enumerate it (Picture 2), but the software didn't recognise my ported device, and it couldn't load "can0" on Ubuntu. So, I used the packet capture method and found that the vendor was missing. The main issue is that on the Ubuntu side, there was no enumeration of any CAN devices, and then I entered dmesg. Hint: [50184.221105] gs_usb 2-2.1:1.0: Couldn't send data format (err=-110). So, can you help me find out what's wrong?

marckleinebudde, 2025-02-20:
Your firmware fails with a timeout https://github.com/torvalds/linux/blob/master/include/uapi/asm-generic/errno.h#L93 when the Linux driver sends the host config https://github.com/torvalds/linux/blob/v6.13/drivers/net/can/usb/gs_usb.c#L1433

There's a problem with your firmware. As you're using a branch that's over 9 years old and you don't show us your code, I don't see any possibilities to help any further.

tonghiden, 2025-02-20:
Oh, thank you. This has been very helpful to me. I'm sorry, I forgot to upload the code because I think the entire project is a bit large and inconvenient, and some parts are encrypted by software. I will try my best to troubleshoot this issue

fenugrec, 2025-02-20:
I recommend you don't waste time with the 1.0 tree for reasons stated above.

Closing as currently out of scope.

closed 2025-02-20 by fenugrec