candle-usb/candleLight_fw #189: Issues porting to STM32405RGT

mowoe, 2025-02-20:
Hi everyone,

i am currently trying to port candlelight to the STM32405RGT
My messy progress can be found here: https://github.com/mowoe/candleLight_fw/tree/feature/STM32F405RGT

After lots of issues, i got some stuff working, the stm correctly enumerates as a gs_usb device. 
But when i try to bring the interface up, the operation times out and the stm gets stuck at https://github.com/mowoe/candleLight_fw/blob/27430416dcdec51a65a46e25d3c703beddd1528e/src/can/bxcan.c#L139
```c
while ((can->MSR & CAN_MSR_INAK) != 0);
```

Unfourtunately, i dont have enough experience with the STM HAL to get what might be wrong here. I largely used the STM32F407 devboard config as a starting ground, but for some things in `config.h` i just dont know what they are for, e.g. `TIM2_CLOCK_SPEED` and `CAN_CLOCK_SPEED`.

I also heavily modified the clock configuration because the stm ran way too slow for some reason with the default setup (i am also using an 8Mhz crystal like on the devboard). I copied this config from a working cubeMX project over and now the time seems okay.

Any help would be appreciated, thanks in advance!

marckleinebudde, 2025-02-21:
`CAN_CLOCK_SPEED` is speed of the CAN core, look at the clock tree in the cubeMX. Aim for a CAN clock multiple of 8MHz.

`TIM2_CLOCK_SPEED` is the speed of the timer2, it's used to divide the timer so that it runs with 1MHz.

mowoe, 2025-02-21:
Thanks @marckleinebudde ! 

The issue of the stm not getting its can interface ready was actually related to this line:
```c
HAL_GPIO_Init(GPIOD, &itd);
```
in `device_f4.c`, which hardcodes on which gpio 'channel' the can interface is. But for the F405, the can channel is at B8 and B9. Changing this fixed this issue and i successfully can bring the interface up. Now i can send can messages and i have verified that other devices on the bus can receive them. However, unfourtunately, the other direction doesnt work yet. The STM does not seem to realise that someone has acknowledged the frame, hence the `gs_usb` driver starts to report `write: No buffer space available` after some time. Also, when receiving a can frame, the STM correctly acknowledges the frame to the bus, but i dont get the frame on my pc.

I unfourtunately dont see a can clock branch in my cubeMX, but through some trial and error with an oscilloscope, i tweaked `CAN_CLOCK_SPEED` so that the bus speed matches what i expect (1Mbit/s). 

I am still unsure how i would come up with the `TIM2_CLOCK_SPEED` value (apart from properly calculating it, which i am not sure how to do). 

But thanks for the clarifications!

marckleinebudde, 2025-02-21:
I think the CAN core is connected to ABP1. This setup should give you 40 MHz on the CAN bus and 160 MHz on the system timers. You effectively lose 8MHz, but get a better CAN clock.

[image]

mowoe, 2025-02-21:
Dont i need the 48MHz clock for the USB stuff? In your screenshot it is set to 80 MHz, which is why it complains in my CubeMX:

[image]

marckleinebudde, 2025-02-21:
yeah right :/, forgot about USB. The clock tree on the newer µC has a dedicated CAN PLL.

mowoe, 2025-02-21:
Thanks, with these values, i can use the same values for `TIM2_CLOCK_SPEED` and `CAN_CLOCK_SPEED` as in the f4 devboard config, to get 1MHz on the bus. 
However the issue with the stm not being able to receive anything (ACKs and full frames) still remains :/