candle-usb/candleLight_fw #190: startup LED blinking might take too ling and lead to timeout of USB init

marckleinebudde, 2025-04-08:
Cc: https://gist.github.com/dickenhobelix

dickenhobelix, 2025-04-08:
I observed this with the multichannel branch at @marckleinebudde on an  STM32F4_DevBoard, when I tried to add a second CAN Channel. 
The host kernel then complains "Device not responding to setup address". I worked around it by pulling the usbd unit before doing the loop with the CAN init. I am not entirely sure if that is the right way ™️, that is why I didn't send a PR yet.

marckleinebudde, 2025-04-08:
I want to rework the LED stuff anyways once I find time for it.

dickenhobelix, 2025-04-08:
@marckleinebudde FTR: I pushed my changes to https://github.com/dickenhobelix/candleLight_fw/tree/dualchannel_stm32f407

fenugrec, 2025-04-08:
Good catch. This fancy LED code has caused subtle issues already in the past. 

I'm not sure why it made a difference to also move the USBD_init call earlier, I thought the host would not even 'detect' the device and start enumeration until we call init...

dickenhobelix, 2025-04-08:
It might be related to the D+ line being tied to 3V3 on my cheap eval board. IIRC this will trigger the enumeration, but the device is not ready yet. But probably someone more knowledgeable in USB should double check this assumption...

fenugrec, 2025-04-08:
@dickenhobelix had you tried just moving the init-blink code after USBD_start() , i.e. outside that for() loop entirely ?

dickenhobelix, 2025-04-08:
> Not yet, but give me a second and I will try it...

blinking after initializing USB does indeed work, at least on my board. Gonna send a PR in due course.