candle-usb/candleLight_fw #195: Firmware restart after USB DFU update fails

m-korber, 2025-04-29:
When updating the firmware on a CandleLightFD adapter the final restart of the newly flashed firmware does not work:

```
mkorber@raspberrypi:~ $ sudo dfu-util -d 1d50:606f -a 0 -s 0x08000000:leave -D candleLightFD_fw_8d3e7fa_2023-07-20.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1d50:606f
Run-Time device DFU version 011a
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
dfu-util: error detaching
Device will detach and reattach...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading element to address = 0x08000000, size = 19504
Erase           [=========================] 100%        19504 bytes
Erase    done.
Download        [=========================] 100%        19504 bytes
Download done.
File downloaded successfully
Submitting leave request...
dfu-util: Error during download get_status
```

After that the USB interface did not automatically re-enumerate but it was required to unplug/replug the adapter.

The issue seems to be very similar to https://github.com/candle-usb/candleLight_fw/pull/176#issuecomment-2128253129 

Any ideas or input on that issue?

fenugrec, 2025-04-29:
once it freezes could you hook up a debug adapter to find out where it is.

marckleinebudde, 2026-09-17:
Closed-by: https://github.com/candle-usb/candleLight_fw/pull/319

closed 2026-09-17 by marckleinebudde