candle-usb/candleLight_fw #196: Testing sending files at 8Mbps

cinlam, 2025-04-30:
Hello everyone,

I am quite new to this, so I may not know the basics very well. For a project, I am trying to send a 2MB file, divided into chunks of 64 bytes, using CANFD. I am using Candlelight FD to achieve this.

To get the highest speed possible, I am sending my data at the maximum data rate of 8 Mbps. I downloaded the latest binary file from the actions section, which fixed a major issue in my code: the transceiver delay. Now I no longer get any "buffer full" errors. Before, I was stuck sending data at 2 Mbps, which is now working fine.

However, when I try to measure the time it takes to send, it's around 5 seconds instead of 2 seconds (if my calculations are correct). I am therefore seeing a data transmission rate of around 3.5 Mbps instead of the expected 8 Mbps.

Does anyone know why this is happening?

Here is the code I use to send my data.

import os
import time
import subprocess
import can

# Default configuration
CAN_CHANNEL = 'can0'
BITRATE = 1_000_000
DBITRATE = 8_000_000
ARBITRATION_ID = 0x123
CHUNK_SIZE = 64
MAX_RETRIES = 10
RETRY_WAIT = 0.001  # seconds

def run_command(command, check=True):
    try:
        subprocess.run(command, check=check, stdout=subprocess.PIPE, stderr=subprocess.PIPE, text=True)
        return True
    except subprocess.CalledProcessError as e:
        print(f"Error while executing command: {' '.join(command)}")
        print(f"Details: {e.stderr.strip()}")
        return False

def is_can_interface_up(channel=CAN_CHANNEL):
    try:
        result = subprocess.run(
            ['ip', '-details', 'link', 'show', channel],
            stdout=subprocess.PIPE,
            stderr=subprocess.PIPE,
            text=True,
            check=True
        )
        return 'state UP' in result.stdout
    except subprocess.CalledProcessError:
        print(f"Error: The interface {channel} does not exist or is not available.")
        return False

def configure_can_interface(channel=CAN_CHANNEL, bitrate=BITRATE, dbitrate=DBITRATE):
    if is_can_interface_up(channel):
        print(f"Interface {channel} is already configured.")
        return True

    print(f"Configuring interface {channel}...")

    if not run_command(['sudo', 'ip', 'link', 'set', channel, 'txqueuelen', '10000']):
        return False

    command = [
        'sudo', 'ip', 'link', 'set', channel, 'up', 'type', 'can',
        'bitrate', str(bitrate), 'dbitrate', str(dbitrage), 'fd', 'on'
    ]
    if run_command(command):
        print(f"Interface {channel} configured with bitrate={bitrate}, dbitrate={dbitrate}")
        return True
    return False

def send_with_retry(bus, msg, retries=MAX_RETRIES, wait=RETRY_WAIT):
    for _ in range(retries):
        try:
            bus.send(msg, timeout=0)
            return True
        except can.CanError:
            time.sleep(wait)
    return False

def send_image_on_can(image_path, channel=CAN_CHANNEL):
    if not os.path.exists(image_path):
        print(f"Image not found: {image_path}")
        return

    with open(image_path, 'rb') as f:
        image_data = f.read()

    total_bytes = len(image_data)
    total_bits = total_bytes * 8

    print(f"Image loaded ({total_bytes} bytes), sending in progress...")

    try:
        bus = can.interface.Bus(channel=channel, interface='socketcan', fd=True)
        total_frames = (total_bytes + CHUNK_SIZE - 1) // CHUNK_SIZE

        start_time = time.perf_counter()
        for i in range(total_frames):
            chunk = image_data[i * CHUNK_SIZE:(i + 1) * CHUNK_SIZE]
            msg = can.Message(
                arbitration_id=ARBITRATION_ID,
                data=chunk,
                is_extended_id=False,
                is_fd=True,
                bitrate_switch=True
            )
            if not send_with_retry(bus, msg):
                print(f"Error: Frame {i + 1}/{total_frames} not sent after several attempts.")
                return
        end_time = time.perf_counter()

        duration = end_time - start_time
        real_bitrate = total_bits / duration
        avg_time_per_frame = duration / total_frames

        print("\n--- Transmission Statistics ---")
        print(f"Total send time: {duration:.6f} seconds")
        print(f"Total size sent: {total_bytes} bytes")
        print(f"Number of frames sent: {total_frames}")
        print(f"Average time per frame: {avg_time_per_frame:.6f} seconds")
        print(f"Observed bitrate: {real_bitrate:.2f} bits/s ({real_bitrate / 1_000_000:.2f} Mbit/s)")

        # End frame
        end_msg = can.Message(
            arbitration_id=ARBITRATION_ID,
            data=b'END',
            is_extended_id=False,
            is_fd=True,
            bitrate_switch=True
        )
        send_with_retry(bus, end_msg)
        print("End frame sent.")

    except Exception as e:
        print(f"CAN Error: {e}")
    finally:
        try:
            bus.shutdown()
            print("CAN interface shut down cleanly.")
        except Exception:
            print("Error while shutting down CAN interface.")

if __name__ == "__main__":
    if configure_can_interface():
        send_image_on_can("image.png")

marckleinebudde, 2025-04-30:
Don't confuse the data bitrate with the effective data rate of the payload.

Can you measure the load of the CAN bus with `canbusload`.

cinlam, 2025-04-30:
canbusload doesn't react when I send the data using my code.

marckleinebudde, 2025-04-30:
use latest the `canbusload` from https://github.com/linux-can/can-utils

cinlam, 2025-04-30:
Ok thank you.
Now ````canbusload```` is working
Here is what I get:

[image]
[image]

cinlam, 2025-04-30:
Ok thank you.
Now canbusload is working
Here is what I get:

[image]
[image]

marckleinebudde, 2025-04-30:
Try `can0@1000000,8000000` to indicate 8Mbit/s data bitrate. Currently you have configured for 1MBit/s only. However I'm not sure if the bus load calculation (`312%` in your screenshots) already takes the arbitration and data bitrate into account.

Are you sending with a CandleLightFD adapter? I think you hit a limitation of the protocol that the CandleLightFD speaks over USB. We have ideas how to improve the TX performance, but it has not been implemented in the firmware not in the Linux driver.

cinlam, 2025-04-30:
You're right, now that I have properly specified the data bitrate, the bus load is around 67%.
[image]

Yes, I'm using the Candlelight FD adapter.
As I understand it, the data bitrate is 8 Mbps (theoretical maximum), but the effective data rate refers to the actual payload throughput, which is lower due to protocol overhead and hardware constraints, is it correct ?

marckleinebudde, 2025-05-03:
Feel free to re-open this or another issue, if you have more questions!

closed 2025-05-03 by marckleinebudde